pikvm/pikvm #291: IR Remote

petrkr, 2021-05-01:
Just keep here for remember me.

Would be handy as future request have IR Remote support (transmitter) so I can use this project for satellite receiver aswell.. I has HDMI output and by splitter it can be used for remote control such receiver.

IR receiver would be handy for record remote control codes.

petrkr, 2021-05-04:
Well some ways changed during time and that article with LIRC. Now it is handled directly by kernel itself and dt overlay.

I made test environment and was able receive and send RC Code

```
[root@pikvm tmp]# ir-keytable -t
Testing events. Please, press CTRL-C to abort.
1315.690030: lirc protocol(nec): scancode = 0xe
1315.690041: event type EV_MSC(0x04): scancode = 0x0e
1315.690041: event type EV_SYN(0x00).
```

```
[root@pikvm ~]# ir-ctl -d /dev/lirc1 -S nec:0xe -v
Sending to kernel encoder protocol:nec scancode:0xe
```

What does not work and do not know why yet is dtoverlay from /boot/config.txt

```
dtoverlay=ir-gpio,gpio_pin=18
dtoverlay=ir-gpio-tx,gpio_pin=10
```
but /dev/ devices are not made. But if I will call dtoverlay by hand from user-space, then it works

```
dtoverlay gpio-ir gpio_pin=18
dtoverlay gpio-ir-tx gpio_pin=10
```

Actually for simulate RC there could be only "tx" part, but "rx" is handy for recording unknown RC for which there is not database, like for every mine Satellite receiver.

strausmann, 2021-07-14:
IR remote support for both sending and receiving would be really super helpful.

My purpose for the IR remote support is to control HDMI KVM switches. In my example I have an Aimos 8 port KVM switch. Unfortunately the HotKeys only work on the port for the keyboard. 
But on this port I can't get any input to the PiKVM to work. Only if the PiKVM is connected to the HUB1 or HUB2 port, the keyboard and mouse control works. But this also makes the HotKey switching impossible.

Currently I have a Yitika IR remote control placed nearby. Using the Android app "Smart Life" I was able to determine the IR signals of the remote control and teach them in the app.

petrkr, 2021-07-14:
I have some idea for UI and data file on backene, but will need somehow integrate it to pikvm's eco system.

In short, i'll take part of  enigma2 remote system, they use for OpenWebIf image of remote and have mapped every button on image, definition is in XML file, so I'll use it as base and extend it by raw scancodes and RC protocol.

mdevaev, 2021-07-17:
This can be implemented as a driver for pseudo-gpio: https://github.com/pikvm/pikvm/blob/master/pages/gpio.md

GralfR, 2022-01-11:
IR would be nice, not only for receivers. I have a 8x8 HDMI-matrix with an IR remote control to switch inputs to outputs.
But for the usecase with receivers, the audio-support (audio input by HDMI) is more important - or at least an analogue audio input or USB-soundcard, but I think audio via HDMI should be the first step.

mdevaev, 2022-05-19:
https://github.com/crutonjohn/pikvm-ir

petrkr, 2022-05-19:
Why use obsolete LIRC and not current kernel based driver which I posted example before?

mdevaev, 2022-05-19:
I don't know. I didn't make this solution.

petrkr, 2022-05-19:
I see now, it's different repo, maybe he did not saw this thread and just tried to make LIRC work, because there are 100+1 howtos on LIRC but nobody will write it's obsolete and should be used kernel driver instead.

That's XY problem, he tries to achieve build of LIRC unless he know he should to use something else to get same result without need to hack anything

crutonjohn, 2022-05-19:
> I see now, it's different repo, maybe he did not saw this thread and just tried to make LIRC work, because there are 100+1 howtos on LIRC but nobody will write it's obsolete and should be used [...]

hey 👋 

yeah, I didn't bother searching for any IR based issues on this repo mainly because I knew it was possible and I wanted to challenge myself to make it work. 

I'd be interested in learning more about these kernel-based drivers that you're referring to. do you have any links to provide so I can get an idea of how it works? I'd much rather use a stable method than relying on LIRC at this point.

petrkr, 2022-05-19:
> I'd be interested in learning more about these kernel-based drivers that you're referring to. do you have any links to provide so I can get an idea of how it works? I'd much rather use a stable method than relying on LIRC at this point.

I do not know how you use it on other parts. But basic you just load kernel module and you can access devices from /dev/lirc... see my example in 2nd comment ( https://github.com/pikvm/pikvm/issues/291#issuecomment-832248866 )

But more examples how to use I do not search those days as that command line tools was enough for me to test control (tx) and learning (rx - decoding).

for TX I have simple IR diode connected on GPIO (with some really low resistor like 100ohm if I remember well, but for pulse it could be without resitor at all, so it's just safe for Rpi's GPIO)

I will try to find that device and post schematics later. Because I do not remember what I've used as receiver if only IR RX diode or already demodulator

crutonjohn, 2022-05-19:
> > I'd be interested in learning more about these kernel-based drivers that you're referring to. do you have any links to provide so I can get an idea of how it works? I'd much rather use a stable method than relying on LIRC at this point.
> 
> I do not know how you use it on other parts. But basic you just load kernel module and you can access devices from /dev/lirc... see my example in 2nd comment ( [#291 (comment)](https://github.com/pikvm/pikvm/issues/291#issuecomment-832248866) )
> 
> But more examples how to use I do not search those days as that command line tools was enough for me to test control (tx) and learning (rx - decoding).
> 
> for TX I have simple IR diode connected on GPIO (with some really low resistor like 100ohm if I remember well, but for pulse it could be without resitor at all, so it's just safe for Rpi's GPIO)
> 
> I will try to find that device and post schematics later. Because I do not remember what I've used as receiver if only IR RX diode or already demodulator

In your post you are using pin 18 for RX and pin 10 for TX.

instead you should try using pin 19 for TX as it is unused by anything PiKVM related, and additionally supports PWM which is required by IR TX operations. I haven't looked very hard into which pin i'm planning on using for RX, but it's looking like 16. 

If you relocate your pins in `/boot/config.txt` that might help because 18 is used by the HDMI sound by default:

> I2S HDMI sound - GPIO 18, 19, 20, 21. Can be used for custom purposes if the tc358743-audio overlay in /boot/config.txt is disabled AND jumpers [4] are removed.

of course, if you've already done that then ignore me.

if i'm understanding you correctly, you're just suggesting that we remove LIRC completely and leverage `ir-ctl` to send/receive? i'm going to do some experimenting on my end to see what the UX looks like. ideally i'd like to move away from LIRC completely because of obvious reasons.

in the interim however, i will be continuing to run the patching of LIRC as defined in my repo as it was the path of least resistance when trying to capture codes from my particular remote. happy to help come up with a long-term and sustainable solution with you though! 

i'll let you know how it shakes out @petrkr 

thanks for bringing this issue to my attention @mdevaev

petrkr, 2022-05-19:
That GPIO was probably from some example, it's already half year ago and those days there was nothing like I2S used HDMI if I remember.

Also it was just suggestion.

Unfortunately I have lot of other open source and free-time projects (read it as projects no one will pay for) to be done. Everything now here goes up and I am aimed more for paid projects. So it was just suggestion for you. If you are OK with patched obsolete LIRC, then I am fine with it :)

mdevaev, 2023-12-03:
Judging by the existing solution, it's pretty hacky, and I must admit that you have done a great job and it is very witty. Although I can't take this to support and I don't think it will ever be done, because this is a extremely rare usecase. I want to close this as wontfix, however, I've added a link to this issue to the official FAQ so that those looking for IR control can use your method. Thank you guys.