pikvm/pikvm #293: Bug in  /etc/kvmd/override.yaml when GPIO pin number is 22 web-server not started

GrafDe, 2021-05-03:
**Describe the bug**
When i try to use pin 22 in "/etc/kvmd/override.yaml"  and restart kvmd i get web-server response "500 Internal Server Error"

# uname -a
Linux pikvm 5.10.23-2-ARCH #1 SMP Thu Mar 18 18:33:35 MSK 2021 armv7l GNU/Linux

this config doesn't work:
#cat /etc/kvmd/override.yaml
`kvmd:
    gpio:
        drivers:
            # This example shows how the default __gpio__ driver settings can be changed. It can be omitted if you are fine wit>
            __gpio__:  # Names surrounded by double underscore are system reserved
                type: gpio  # Refers to the plugin name handling the communication
        scheme:
            # Two outputs of RPi's GPIO
            button1:
                pin: 4 # GPIO pin number on the RPi
                mode: output
                #switch: false  # Disable switching, only pulse available
            button2:
                pin: 22 # GPIO pin number on the RPi
                mode: output
                #switch: false # Disable switching, only pulse available
            button3:
                pin: 6 # GPIO pin number on the RPi
                mode: output
                #switch: false  # Disable switching, only pulse available
            button4:
                pin: 26 # GPIO pin number on the RPi
                mode: output
                #switch: false # Disable switching, only pulse available
        view:
            header:
                title: Power Switches  # the menu title
            table:  # The menu items are rendered in the form of a table of text labels and controls
                - ["#Press Power Buttons"]  # Text starting with the sharp symbol will be a label
                - []  # creates a horizontal separator and starts a new table
                - ["#Comp 1:", button1]  # Text label, one input, one button with text "Click"
                - ["#Comp 2:", button2]
                - ["#Comp 3:", button3]
                - ["#Comp 4:", button4]
                - []  # creates a horizontal separator and starts a new table`

and when i delete button2 with pin 22 all works fine. or i changed pin to another of 22 and all work fine.
#cat /etc/kvmd/override.yaml
`kvmd:
    gpio:
        drivers:
            # This example shows how the default __gpio__ driver settings can be changed. It can be omitted if you are fine wit>
            __gpio__:  # Names surrounded by double underscore are system reserved
                type: gpio  # Refers to the plugin name handling the communication
        scheme:
            # Two outputs of RPi's GPIO
            button1:
                pin: 4 # GPIO pin number on the RPi
                mode: output
                #switch: false  # Disable switching, only pulse available
            button3:
                pin: 6 # GPIO pin number on the RPi
                mode: output
                #switch: false  # Disable switching, only pulse available
            button4:
                pin: 26 # GPIO pin number on the RPi
                mode: output
                #switch: false # Disable switching, only pulse available
        view:
            header:
                title: Power Switches  # the menu title
            table:  # The menu items are rendered in the form of a table of text labels and controls
                - ["#Press Power Buttons"]  # Text starting with the sharp symbol will be a label
                - []  # creates a horizontal separator and starts a new table
                - ["#Comp 1:", button1]  # Text label, one input, one button with text "Click"
                - ["#Comp 3:", button3]
                - ["#Comp 4:", button4]
                - []  # creates a horizontal separator and starts a new table`

I need use pin 22 because my relay board is use it
link to my relay module https://wiki.keyestudio.com/KS0212_keyestudio_RPI_4-channel_Relay_Shield?spm=a2g0s.imconversation.0.0.1c573e5fYFHbXJ

mdevaev, 2021-05-03:
This is not a bug. Pin 22 is used for the HDD activity indicator. If you look to `journalctl -u kvmd`, you will see that kvmd cannot start because your configuration conflicts with the default settings. To use pin 22, you need to move the ATX function to another location (or disable it altogether) using override:
```yaml
kvmd:
    atx:
        hdd_led_pin: N
```
Or:
```yaml
kvmd:
    atx:
        type: disabled
```
Use `kvmd -m` to view entire config.

GrafDe, 2021-05-03:
tanks, is work
i was change in /etc/kvmd/main.yaml 
```
kvmd:
    atx:
        type: disabled
```

mdevaev, 2021-05-03:
Welcome

closed 2021-05-03 by mdevaev