pikvm/pikvm #294: what to enter in the override.yaml to make the relay work

GrafDe, 2021-05-03:
i have usb-relay. it is detected as 
```
# lsusb
Bus 001 Device 003: ID 5131:2007 MSR MSR-101U Mini HID magnetic card reader
...
# ls /dev/hid*
/dev/hidraw0
```
to close the relay, I send the command in console
```
printf '\xFF\x55' >/dev/hidraw0
```
1) How i can send a signal to close the relay through web-interface ? what i need write in /etc/kvmd/override.yaml

2) I am currently using a bash script from the console to close this relay. This is inconvenient because i have to open the console. Is there a way to run bash scripts from the web interface, like writing multiple lines to /etc/kvmd/override.yaml?

strausmann, 2021-05-03:
This works for me, after a long after a long fiddling :-)

```yaml
kvmd:
    gpio:
        drivers:
            relay:
                type: hidrelay
                device: /dev/hidraw0
        scheme:
            relay1:
                driver: relay
                pin: 0
                mode: output
                initial: null
                switch: false
            relay2:
                driver: relay
                pin: 1
                mode: output
                initial: null
                switch: false
            relay3:
                driver: relay
                pin: 2
                mode: output
                initial: null
                switch: false
            relay4:
                driver: relay
                pin: 3
                mode: output
                initial: null
                switch: false
        view:
            header:
               title: "HDMI Switch"
            table:
                - ["#HID Relays /dev/hidraw0"]
                - []
                - ["#Port #1:", "relay1"]
                - ["#Port #2:", "relay2"]
                - ["#Port #3:", "relay3"]
                - ["#Port #4:", "relay4"]
```

GrafDe, 2021-05-04:
Thanks for you answer, strausmann. But this not work for my relay, i try this way before send my post yesterday. for my relay work i need sell to relay two byte \xFF and \x55 . I wrote about this in the first post) How it is doing from web?

arch1mede, 2021-05-04:
@GrafDe gpio for the relay is only supported at this time and ustom buttons/scripts is not supported at this time, maybe in the future.
For now, you'll have to contend with clicking the web term button and running a script, or ssh'ing in and running a script. Some ssh clients allow you to run scripts on login.

mdevaev, 2021-05-06:
@GrafDe Try to use pin 54, pin 55 nd so on in the `scheme` section. I think it would help.

mdevaev, 2021-05-22:
I think the problem is solved. Write, if this is not the case - we will continue.

closed 2021-05-22 by mdevaev

byrnes-patrick, 2024-05-11:
I tried the example by @strausmann without success and tried changing the pins like @mdevaev said and the pikvm service won't start.

JeWe37, 2024-05-25:
So I ran into this and from what I can tell it is not fixed. These relays use an entirely different protocol from the supported kind. 

The protocol seems to be similar to relays of the type LCUS-{1,2,4,8} based on USB-serial chips, but use HID instead. Thus they show up as hidraw devices rather than ttys. Notably lacking though seems to be the ability to get the current state of the relay.

I will probably submit a PR with support for these, it should be pretty simple to implement by just directly writing to the char dev.

mdevaev, 2024-05-25:
Hekko. Use `noyito` driver instead of `hidrelay`. Note that they are not capable of synchronizing states, so when you restart pikvm/kvmd, they will be reset.

Separately, I want to say that these relays are a piece of shit, because there was no point in making them hid at all if their manufacturer could not make a status response command. Not to mention that it is possible even for a serial port.

If you have the opportunity, return it to the seller.

JeWe37, 2024-05-25:
>  Hello. Please use noyito driver instead of hidrelay. Note that they are not capable of synchronizing states, so when you restart pikvm/kvmd, they will be reset.

Ah, I hadn't seen that yet and my version of PiKVM wasn't new enough, now I know what @byrnes-patrick was referring to.

> Separately, I want to say that these relays are a piece of crap, because there was no point in making them hid at all if their manufacturer could not make a status response command. Not to mention that it is possible even for a serial port.
> 
> If you have the opportunity, return it to the seller.

Returns and Chinese sellers are rarely fun, but undoubtedly there are better relays, I'd just rather want a solution now than later.