pikvm/pikvm #295: Hello, how to emulate the usb interface of raspberrypi3a as a mass storage device.

wnark, 2021-05-05:
I use the v0 solution, and the raspberrypi3B method is successful on raspberrypi3a.
[image]
I found that when using a mobile power bank and the same power cord, raspberrypi3B will report that the voltage is insufficient, while raspberrypi3a is normal。
Now there is a question, how to manually add an emulated mass storage device to it on the web side?
[image]

And I found that the v0 version does not seem to have automatic partition expansion.
[image]
~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
The usb-a of the Raspberry Pi 3A seems to be able to be simulated as a large-capacity device, although I am not sure how to implement it

https://hexdocs.pm/nerves_system_rpi3a/readme.html

https://gist.github.com/gbaman/50b6cca61dd1c3f88f41

~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
https://github.com/pikvm/pikvm/issues/158

If you need rj45, you can use this method

~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~

wnark, 2021-05-05:
I was wondering, can I use raspberrypi4B mirroring directly on raspberrypi3A, and then modify a certain configuration file, modify the USB ID?, convert usbc into a large-capacity device, and convert usba into a large-capacity device.
Since Raspberry Pi 3a and 4a are both arm, although their architecture is different, they may be compatible. (It’s just that I don’t know which file to modify) lol

wnark, 2021-05-05:
There is another question. Since the RAM of the Raspberry Pi 3a is only 512M, maybe I change the root directory to be writable, and then I can mount the swap partition in a normal Linux way? Or use the unallocated partition of the tf card as the swap partition?
Can this improve the problem of long-time use of the screen and operation freeze?

wnark, 2021-05-05:
Very strange, it was very smooth at first, but it didn't take long for the keyboard and mouse to react very slowly, and the picture dropped from 30 to 2 frames.
At first I thought it was the reason for the lack of memory. It seems that compatibility is still not good somewhere?

mdevaev, 2021-05-09:
Sorry, I understand nothing. What are you trying to do?

wnark, 2021-06-23:
[image]
[image]
It is not recommended to use Raspberry Pi 3A. The internal bus transmission speed of Raspberry Pi 3A may be only 14Mb, which will not fully utilize the performance of Raspberry Pi 3A.

wnark, 2021-06-23:
I was planning to use the USB port of the Raspberry Pi 3A to simulate the device, so I did not use the USB network card. It seems that using w5500 or wifi is too slow and not suitable for use, so the plan to use Raspberry Pi 3A instead of Raspberry Pi 4B failed.

wnark, 2021-06-23:
If the USB port is occupied in order to use the USB network card, it is better to use Raspberry Pi 2b, which costs less

wnark, 2021-06-23:
```
[root@ipmi ~]# lsusb
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```
emmm,The USB port of the Raspberry Pi 3A is not directly connected to the chip, and it cannot replace the Raspberry Pi 4B.

mdevaev, 2021-07-04:
I have never held this device in my hands and I don't know what to tell you, sorry :(

mdevaev, 2021-07-24:
I suppose I can close it. This is interesting information, but no work is expected here.

closed 2021-07-24 by mdevaev