pikvm/pikvm #296: USB Ethernet Gadget Address Issue

adjscent, 2021-05-05:
**Describe the bug**
Ubuntu 20.04.1, when connected to the pikvm via USB-Ethernet Gadget, does not forward packets properly as it seems to not like 169.254.X.X addresses. Propose to be able to configure the IP address of usb0 and its dhcp range.

**To Reproduce**
Steps to reproduce the behaviour, like:
1. Connect PiKvm to a pc running Ubuntu 20.04.1
2. Run ```ping 8.8.8.8``` on the pc. 
3. Ping fails
4. Tcpdump shows that the pc did not use the source address 169.254.0.X given by the pikvm's dhcp server.

**Expected behaviour**
Ping should work.

**Desktop (please complete the following information):**
 - OS: Ubuntu 20.04.1

**Pi-KVM info:**
  - Raspberry Pi board version [RPi 4]
  - Pi-KVM platform [v2-hdmi]
  - Video capture type [USB]
  - KVMD version: kvmd 2.62-1 kvmd-platform-v2-hdmiusb-rpi4 2.62-1 kvmd-webterm 0.38-1
  - uStreamer version: ustreamer 3.27-1
  - Linux kernel: Linux pikvm 5.10.33-1-ARCH #1 SMP Fri Apr 30 17:52:59 UTC 2021 armv7l GNU/Linux

**Additional context**
The USB Ethernet works well on my Windows 10 20H2 pcs.

arch1mede, 2021-05-09:
The "bug" is detailed in step 2 of https://github.com/pikvm/pikvm/blob/master/pages/usb_ethernet.md#ethernet-over-usb-network, this function is for communication between pikvm and target ONLY

If you are looking at eth pass through you might be interested in: https://pastebin.com/u/srepac override.yaml

adjscent, 2021-05-11:
My issue is with the DHCP range of the otgnet. But I realise it is a TODO feature so I will close this.

mdevaev, 2021-05-14:
@adjscent You can customize internal usb network range in `/etc/kvmd/override.yaml`. For example:
```yaml
otgnet:
    iface:
        net: 10.0.0.0/28
```