pikvm/pikvm #297: A special Raspberry Pi development board-raspberrypi 3B MINI

wnark, 2021-05-05:
In the Chinese second-hand market, there is also a magical board, the Raspberry Pi 3B mini, which is customized by the Raspberry Pi Foundation for htc. The advantage is that it is smaller than the Raspberry Pi 3A, has lower power consumption, and has memory Still 1G, with csi interface! ! !
The disadvantage is that the USB port has a HUB chip, which cannot be used as an analog device, but two USB-A ports are reserved (although welding is required)
Maybe you can take a look at this video (emmm, in Chinese, maybe you need to use Google Translate)
In China, it is used as a model aircraft development board. It has mature gameplay. If you are interested, you can try to buy one for a try?

https://www.bilibili.com/video/BV1Ho4y197ZK

wnark, 2021-05-05:
The shortcomings of no WiFi can be solved with USB wifi/w5500

mdevaev, 2021-05-09:
Sorry, I don't have any extra money right now to test modules other than upstream raspberry.

closed 2021-05-20 by mdevaev