pikvm/pikvm #298: Cannot connect to host computer from RPI Zero W

scottnapleton, 2021-05-08:
**Describe the bug**
I have RPI Zero W connected to Geekworm HDMI to CSI-2 (https://www.amazon.com/gp/products/B0899L6ZXZ/). I'm up to the step where I have connected to the Pi from a client computer and I get the following screen:
[image]

I have verified that the host computer is producing HDMI output and the Module recognizes that there is video attached. Also the device manager recognizes the the mouse and keyboard being emulated by the Pi Zero W.

Here is a picture of my setup:
[image]

I've tried flashing and formatting the SD card multiple times. I have also swapped out numerous USB cables and tried adjusting the ribbon between the Pi and Geekworm module numerous times. I've run out of ideas at this point. Any help is greatly appreciated.

mdevaev, 2021-05-08:
Which browser do you use?

arch1mede, 2021-05-09:
I tried to assist this person and he logged off the server without a proper response, I would recommend this issue be closed as this appears to be due to something on the OS environment at play since he gets the same thing using Chrome/Incog and FF/Private window

scottnapleton, 2021-05-09:
I just tried using a different client computer and was able to connect with no problem. I cannot figure out why the first client was unsuccessful but the issue must lie somewhere on my end.

mdevaev, 2021-05-09:
Okay.

closed 2021-05-09 by mdevaev