pikvm/pikvm #299: REQUEST - Proper changelog

arch1mede, 2021-05-09:
Please add a proper changelog for this project. Its very difficult to look at what each aspect of this projects changes are

Like:
```
2.63                    kvmd
    - Blah
0.1                     kvmd-fan
    - Blah
0.3                     kvmd-oled
    - Blah
2.63                    kvmd-platform-v2-hdmiusb-rpi4
    - Blah
0.39                    kvmd-webterm
    - Blah
3.27                    ustreamer
    - Blah
```

mdevaev, 2021-05-20:
Someone will have to rewrite the news from Discord in the ChangeLog format

sataris, 2022-04-28:
This would be incredibly useful, I'm sure I'm not the only one who has no interest in discord.

mdevaev, 2025-07-25:
The most important release notes was moved from discord to the blog: https://docs.pikvm.org/blog/