pikvm/pikvm #302: Set some default otg-net and firewall parameters on Pi Zero, to avoid having inaccessible unit

bozserz, 2021-05-10:
On Pi Zero WiFi is the only network now. If connection is not available, then we loose access to it.
In some of the Arch Linux distributions the WiFi firmware maybe broken, so after an upgrade WiFi may disappear.
Either because of this or if the WiFi SSID is being hidden, or just because of a typo in the setup, it may happen, that the Pi Zero will be unavailable via the wireless network, and maybe the complete installation should be started over.

If the otg-net (and the firewall) would be set up in any way as default, then the unit would be still available via this connection, even if wireless is not working, and may fix the issues. 
On Windows e.g. the RNDIS  over usb driver is there and could be installed manually (however it's not automatic), so it will be possible to connect immediately.

An alternative is to enable the serial console as default and give console access, but I think the otg-net is easier to use and maybe would be used by more people anyway. 
It's just an idea (I had issues myself, so maybe it could save some time for the others too).

Regards: Zoltan

arch1mede, 2021-06-10:
I feel this should be up to the end user to ensure accessibility throughout different network scenarios

mdevaev, 2021-07-04:
Do you have any suggestions on how this should be done? Maybe something like boot config as for wifi?

arch1mede, 2021-07-04:
> 
> 
> Do you have any suggestions on how this should be done? Maybe something like boot config as for wifi?

was this directed at me? I do not, I still maintain that this needs to be setup by the end user as this falls under the Arch linux support over pikvm.

If its a matter of accessing the zerow due to "some error" then I would recommend using [this](https://www.pishop.us/product/solderless-serial-to-usb-adapter-for-raspberry-pi-zero/)

mdevaev, 2021-11-08:
In new images, you can add the `ENABLE_ORG_SERIAL=1` parameter to `/boot/pikvm.txt` (the first partition of the memory card) to activate the OTG network, so that you can get to the OS.

Firewall setup is a huge and immense topic, and creating a configurator that would suit all users is clearly beyond the goals of PiKVM. As an emergency means, `ENABLE_OTG_SERIAL` is more than enough, for the rest of the cases, use any of the iptables configuration tools.

closed 2021-11-08 by mdevaev