pikvm/pikvm #306: Pasting into KVM target results in incorrect characters

sIKE23, 2021-05-12:
Copy a 9 line script from Notepad++ to the Paste text as a keypress sequence results in bad characters/missing spaces etc:

$value = cmd /c  ‘wmic service get name,displayname,pathname,startmode |findstr /i “auto” |findstr /i /v “c:windows\” |findstr /i /v “””‘
if ($value -like “* *”)
   {
   1
   }
   else
   {
   0
   }

[image]
 

You will note that the single quote preceding wmic is missing and the open parentheses is a 9 instead. 

I am seeing this allot today but have seen it since the update that added in the paste toggle widget.

mdevaev, 2021-05-14:
I'll check it

mdevaev, 2021-05-16:
As far as I can see, notepad++ for some reason uses not ascii quotes, but typographic ones. Even if you can insert this into the terminal, the script will not work.

I am more interested in why there is no opening parenthesis and where the $ character disappeared. If you repeatedly repeat the insertion, will these characters be corrupted each time?

sIKE23, 2021-05-16:
I had the same thought, and I tried it multiple times, each time it corrupted the paste but not necessarily is the same locations. As an example, the "Open Parenthesis" would paste properly other times it was the numeral 9. 

If I am not mistaken, Notepad ++ by default is UTF-8.

mdevaev, 2021-05-19:
That's the reason. Instead of ASCII quotes, Notepad++ uses quotes from the localization.

sIKE23, 2021-05-19:
The other thing I uncovered is that when copying from a web page a dash on the web page is considered a Em Dash  which is not available in ascii. So my guess is that the keyboard 'typer" is just dropping the long dash.

mdevaev, 2021-05-19:
I'll try to fix it somehow.

sIKE23, 2021-05-19:
I think you will need to convert the pasting code to UTF-8 since you have added localization support.

mdevaev, 2021-05-19:
Yes: https://github.com/pikvm/kvmd/blob/master/kvmd/keyboard/printer.py#L44

mdevaev, 2021-05-19:
I installed myself notepad++ and don't see this problem with quotes - it uses ascii quotes. even if I remove the restrictions from the input, these quotes cannot be entered from the keyboard, because they are typographic quotes. What is your localization?

sIKE23, 2021-05-19:
US-EN 

I am a sysadmin. So I spend a good amount of time doing Google searches and pulling bits and pieces (scripts/commands) from web sites and then pasting them into my Laptop via the KVM.

sIKE23, 2021-05-19:
Just to be clear that was a copy from a website, with a paste into Notepad++ on the Laptop from the KVM Paste.

mdevaev, 2021-05-20:
Is this a correct result for the text?
```
$value = cmd /c 'wmic service get name,displayname,pathname,startmode |findstr /i "auto" |findstr /i /v "c:windows\" |findstr /i /v """'
if ($value -like "* *")
{
1
}
else
{
0
}
```

There's something strange with the quotation marks at the end of the first line in the original and inserted text. Is this how it should be?

sIKE23, 2021-05-20:
yep

mdevaev, 2021-05-20:
Okay, then take these files and place it on your device:
https://raw.githubusercontent.com/pikvm/kvmd/master/kvmd/keyboard/printer.py to `/usr/lib/python3.9/site-packages/kvmd/keyboard/`
https://raw.githubusercontent.com/pikvm/kvmd/master/web/share/js/kvm/hid.js to `/usr/share/kvmd/web/share/js/kvm/`
Run `systemctl restart kvmd` and check paste funciton. If everything is in order, I will include it in the release.

sIKE23, 2021-05-20:
I did as requested, and everything is working as expected! You are  a rock star.  Here is a link to the page where I got that scriptlet from. 

https://sccmf12twice.com/2017/05/remediate-unquoted-service-path-vulnerability-via-compliance-item/

Let me run with this tomorrow during work at I am constantly running copy paste through the KVM to the target device(s). I will let you know if I see anything else wonky.

mdevaev, 2021-05-20:
Great! I hope that everything will be fine.

mdevaev, 2021-05-21:
I will close this issue, If something doesn't work well, let me know and I'll open it again.

closed 2021-05-21 by mdevaev