pikvm/pikvm #308: bluetooth hid server crashing when starting up.

abh, 2021-05-12:
**Describe the bug**
bt hid server crashing when starting up.

**To Reproduce**
Follow bluetooth hid instructions and reboot.

**Expected behavior**
RPI ready to pair.

**Screenshots**
```
May 12 06:44:02 pikvm kvmd[497]: kvmd.plugins.hid.bt.server        INFO --- Publishing ...
May 12 06:44:02 pikvm kvmd[497]: kvmd.plugins.hid.bt.server        INFO --- Listening [E4:5F:01:22:73:B5]:17 for CTL ...
May 12 06:44:02 pikvm kvmd[497]: kvmd.plugins.hid.bt.server        INFO --- Unpublishing ...
May 12 06:44:02 pikvm kvmd[497]: kvmd.plugins.hid.bt              ERROR --- Unexpected HID error
May 12 06:44:02 pikvm kvmd[497]: Traceback (most recent call last):
May 12 06:44:02 pikvm kvmd[497]:   File "/usr/lib/python3.9/site-packages/kvmd/plugins/hid/bt/__init__.py", line 200, in __server_worker
May 12 06:44:02 pikvm kvmd[497]:     self.__server.run()
May 12 06:44:02 pikvm kvmd[497]:   File "/usr/lib/python3.9/site-packages/kvmd/plugins/hid/bt/server.py", line 128, in run
May 12 06:44:02 pikvm kvmd[497]:     with self.__listen("CTL", addr, HID_CTL_PORT) as ctl_sock:
May 12 06:44:02 pikvm kvmd[497]:   File "/usr/lib/python3.9/contextlib.py", line 117, in __enter__
May 12 06:44:02 pikvm kvmd[497]:     return next(self.gen)
May 12 06:44:02 pikvm kvmd[497]:   File "/usr/lib/python3.9/site-packages/kvmd/plugins/hid/bt/server.py", line 158, in __listen
May 12 06:44:02 pikvm kvmd[497]:     sock.bind((addr, port))
May 12 06:44:02 pikvm kvmd[497]: OSError: bind(): bad family
```

**Pi-KVM info:**
  - RPi 4
  - v2-hdmi
  - USB HDMI
  - KVMD version: kvmd 2.65-1 kvmd-fan 0.1-1 kvmd-oled 0.3-1 kvmd-platform-v2-hdmiusb-rpi4 2.65-1 kvmd-webterm 0.39-1
  - Linux kernel: 5.10.23-2-ARCH

mdevaev, 2021-05-12:
https://bugs.archlinux.org/task/70438

mdevaev, 2021-05-14:
Please update your system and try again.

mdevaev, 2021-05-16:
@abh boop

mdevaev, 2021-05-17:
Fixed with update.

closed 2021-05-17 by mdevaev

abh, 2021-05-18:
thank you @mdevaev  -- I wanted to wait until I'd actually set it up to confirm, but the errors in the logs are indeed gone after an update!