pikvm/pikvm #313: Error occured on HidRelay(relay) on /dev/hidraw0 while reading: IndexError: list index out of range

depeo, 2021-05-14:
Hi,

Trying to get my usb relay working, it works fine via the usbrelay program.

`[root@pikvm(g:master) g:/]# ./usbrelay  -d
Version: 0.7-40-g6d7d664718
Library Version: 0.7-40-g6d7d664718
enumerate_relay_boards()Found 1 devices
Device Found
  type: 0519 2018
  path: /dev/hidraw0
  serial_number: A0001
Manufacturer: Ucreatefun.com
  Product:      HIDRelay
  Release:      1
  Interface:    0
  Number of Relays = 9
  Module_type = 2
A0001_1=-1
A0001_2=-1
A0001_3=-1
A0001_4=-1
A0001_5=-1
A0001_6=-1
A0001_7=-1
A0001_8=-1
A0001_9=-1`

I can toggle the relay correctly by running,
`./usbrelay A0001_2=1
sleep 0.3
./usbrelay A0001_2=0`

udev permissions is set:
`[root@pikvm(g:master) g:/]# ls -ltr /dev/hidraw0
crw-rw-rw- 1 root root 243, 0 May 11 02:19 /dev/hidraw0`

My override.yaml looks like this:
`kvmd:
    gpio:
        drivers:
            relay:
                type: hidrelay
                device: /dev/hidraw0
        scheme:
            relay1:
                driver: relay
                pin: 0
                mode: output
                switch: false
                initial: null
            relay2:
                driver: relay
                pin: 1
                mode: output
                switch: false
                initial: null

        view:
            header:
                title: Switches
            table:
                - ["#HID Relays /dev/hidraw0"]
                - []
                - ["#Relay #1:", "relay1"]
                - ["#Relay #2:", "relay2"]`
            

kvmd daemon log is full of this:
`May 14 18:38:16 pikvm kvmd[370]: kvmd.plugins.ugpio.hidrelay       INFO --- Opened HidRelay(relay) on /dev/hidraw0 while reading
May 14 18:38:16 pikvm kvmd[370]: kvmd.plugins.ugpio.hidrelay      ERROR --- Error occured on HidRelay(relay) on /dev/hidraw0 while reading: IndexError: list index out of range
May 14 18:38:16 pikvm kvmd[370]: kvmd.plugins.ugpio.hidrelay       INFO --- Closed HidRelay(relay) on /dev/hidraw0`

Any ideas how to get it working from the web ui?

mdevaev, 2021-05-14:
Contact me via discord

mdevaev, 2021-05-14:
The bottom line: the relay has a strange firmware that does not support the command to get the channel states. This can't be fixed on our side.

closed 2021-05-14 by mdevaev