pikvm/pikvm #315: keyboard free inactive/busy - no keyboard/mouse in web GUI

sharicov, 2021-05-19:
zero pi w
pikvm updated to the latest version
in web GUI keyboard and mouse in orange.
When hoovering over with mouse says "keyboard free inactive/busy)

dtoverlay=dwc2,dr_mode=peripheral ; not commented

Please assist

sharicov, 2021-05-19:
journalctl -u kvmd
-- Journal begins at Thu 2021-04-08 16:15:06 MSK, ends at Wed 2021-05-19 21:30:33 MSK. --
Apr 08 16:15:34 pikvm systemd[1]: Started Pi-KVM - The main daemon.
Apr 08 16:15:57 pikvm kvmd[356]: Config error: The node 'vnc' must be a dictionary
Apr 08 16:15:58 pikvm systemd[1]: kvmd.service: Main process exited, code=exited, status=1/FAILURE
Apr 08 16:16:08 pikvm kvmd-cleanup[364]: Config error: The node 'vnc' must be a dictionary
Apr 08 16:16:08 pikvm systemd[1]: kvmd.service: State 'stop-post' timed out. Terminating.
Apr 08 16:16:08 pikvm systemd[1]: kvmd.service: Control process exited, code=exited, status=1/FAILURE
Apr 08 16:16:08 pikvm systemd[1]: kvmd.service: Failed with result 'exit-code'.
Apr 08 16:16:08 pikvm systemd[1]: kvmd.service: Consumed 16.509s CPU time.
May 19 18:49:53 pikvm systemd[1]: kvmd.service: Scheduled restart job, restart counter is at 1.
May 19 18:49:53 pikvm systemd[1]: Stopped Pi-KVM - The main daemon.
May 19 18:49:53 pikvm systemd[1]: kvmd.service: Consumed 16.509s CPU time.
May 19 18:49:53 pikvm systemd[1]: Started Pi-KVM - The main daemon.
May 19 18:50:13 pikvm kvmd[381]: Config error: The node 'vnc' must be a dictionary
May 19 18:50:14 pikvm systemd[1]: kvmd.service: Main process exited, code=exited, status=1/FAILURE
May 19 18:50:24 pikvm systemd[1]: kvmd.service: State 'stop-post' timed out. Terminating.
May 19 18:50:24 pikvm systemd[1]: kvmd.service: Control process exited, code=killed, status=15/TERM
May 19 18:50:24 pikvm systemd[1]: kvmd.service: Failed with result 'exit-code'.
May 19 18:50:24 pikvm systemd[1]: kvmd.service: Consumed 15.517s CPU time.
May 19 18:50:27 pikvm systemd[1]: kvmd.service: Scheduled restart job, restart counter is at 2.
May 19 18:50:27 pikvm systemd[1]: Stopped Pi-KVM - The main daemon.
May 19 18:50:27 pikvm systemd[1]: kvmd.service: Consumed 15.517s CPU time.
May 19 18:50:27 pikvm systemd[1]: Started Pi-KVM - The main daemon.
May 19 18:50:46 pikvm kvmd[432]: Config error: The node 'vnc' must be a dictionary
May 19 18:50:47 pikvm systemd[1]: kvmd.service: Main process exited, code=exited, status=1/FAILURE
May 19 18:50:55 pikvm kvmd-cleanup[435]: Config error: The node 'vnc' must be a dictionary
May 19 18:50:56 pikvm systemd[1]: kvmd.service: Control process exited, code=exited, status=1/FAILURE
May 19 18:50:56 pikvm systemd[1]: kvmd.service: Failed with result 'exit-code'.
May 19 18:50:56 pikvm systemd[1]: kvmd.service: Consumed 16.317s CPU time.

mdevaev, 2021-05-19:
You have a syntax error in /etc/kvmd/override.yaml

sharicov, 2021-05-20:
Thanks for quick response, Maxim.
Commenting all the changes and rebooting fixed the problem.

closed 2021-05-21 by mdevaev

Ammara5251, 2021-12-09:
I am trying and have the same problem. How did you resolve it?

sharicov, 2021-12-09:
As far as I remember I just undo all the changes I did to  /etc/kvmd/override.yaml and it started working.

sharicov, 2021-12-09:
If it's VNC related, that's what I have in my records:

Enable VNC (use BVNC client for android):

Switch file system to read/write:

rw
Uncomment in /etc/kvmd/override.yaml

vnc: 
    auth: 
        vncauth: 
            enabled: true 
To set passphrases edit file /etc/kvmd/vncpasswd.

Enable kvmd-vnc daemon. VNC will be available on the port 5900:

systemctl enable --now kvmd-vnc
Switch filesystem back to readonly:

ro

Ammara5251, 2021-12-09:
Thanks for your quick response. So for now, I am just using splitter cable and using hdmi usb dongle. I powered up raspberry pi and logged into pikvm. Now when I use KVM stream, I cannot use mouse or keyboard. I am not using vnc as of now.

[root@pikvm kvmd-webterm]# more /etc/kvmd/override.yaml 
####################################################################
#                                                                  #
# Override Pi-KVM system settings. This file uses the YAML syntax. #
#                                                                  #
#    https://github.com/pikvm/pikvm/blob/master/pages/config.md    #
#                                                                  #
# All overridden parameters will be applied AFTER other configs    #
# and "!include" directives and BEFORE validation.                 #
# Not: Sections should be combined under shared keys.              #
#                                                                  #
####################################################################
#
#########
# Wrong #
#########
#kvmd:
#    gpio:
#        drivers: ...
#kvmd:
#    gpio:
#        scheme: ...
#
###########
# Correct #
###########
#kvmd:
#    gpio:
#        drivers: ...
#        scheme: ...
#
###########
# Example #
###########
#vnc:
#    # See https://github.com/pikvm/pikvm/blob/master/pages/vnc.md
#    keymap: /usr/share/kvmd/keymaps/ru # Set russian keymap
#    auth:
#        vncauth:
#            enabled: true # Enable auth via /etc/kvmd/vncpasswd
kvmd:
    hid:
        mouse_alt:
            device: /dev/kvmd-hid-mouse-alt
#        

#kvmd:
#    msd:
#        type: disabled

#kvmd:
#    hid:
#        mouse:
#            absolute: false
#            horizontal_wheel: false

[image]

sharicov, 2021-12-09:
Physically connected mouse and keyboard will not work since raspberry is in
the USB gadget mode.

Ammara5251, 2021-12-09:
Sorry might be a stupid question, but after connecting the usb cable A to the physical machine without 5V power (its been disabled), shouldn't it use it to control mouse?

[image]

Ammara5251, 2021-12-09:
https://www.tomshardware.com/how-to/kvm-over-ip-raspberry-pi

I am referring to this for building my pikvm.

sharicov, 2021-12-09:
Looks like a very comprehensive guide, however it is not mentioning
connecting physical mouse or keyboard.

mitch98, 2023-03-05:
I'm having the same issue

sharicov, 2025-10-26:
unfortunately I don't remember how i fixed it, was long time ago