pikvm/pikvm #316: 'make os' failed - hkps keyserver not reachable

QEMU90, 2021-05-20:
**Describe the bug**

Building the image failed... I started the building process with the command "make os" and after a while i get the following error:

```
...
gpg: keyserver receive failed: No data
==> EROR: Remote key not fetched correctly from keyserver.
gpg: keyserver receive failed: No data
==> EROR: Remote key not fetched correctly from keyserver.
...

```
As i understand it, the script tries to reach the pool of keyserver "hkps.pool.sks-keyservers.net" and can't reach it.

https://sks-keyservers.net/status/ shows, that **hkps** isn't supported anymore...:

"This service is deprecated. This means it is no longer maintained, and new HKPS certificates will not be issued. Service reliability should not be expected."

**To Reproduce**
Steps to reproduce the behavior, like:
1. Go to 'Building the OS'
2. Finish the steps 0 to 3
3. Scroll down to '4. Build the OS... -> make os'
4. See error

**Desktop:**
 - OS: Ubuntu 20.04 and Manjaro 20.2.1

mdevaev, 2021-05-21:
sks-keyservers.net is dead now. I've replaced it. Run `cd .pi-builder && git pull --rebase && cd ..` and run `make os` again.

mdevaev, 2021-05-21:
Let me know if you encounter eny problems.

QEMU90, 2021-05-26:
No difference after `git pull --rebase`. Also tried on a fresh installed operating system. Still getting the same error.
[image]

wnark, 2021-05-27:
When I was compiling the Raspberry Pi 3B version, although the compilation was completed, the obtained image failed to start. I doubt that the compiled image is incomplete for this reason.
https://mega.nz/folder/9ppWjTqB#t4TdX3cm5MqNmb9lmAdO3Q

wnark, 2021-05-27:
> When I was compiling the Raspberry Pi 3B version, although the compilation was completed, the obtained image failed to start. I doubt that the compiled image is incomplete for this reason.
> https://mega.nz/folder/9ppWjTqB#t4TdX3cm5MqNmb9lmAdO3Q

This link is the newly generated image
The root and admin password is cckexand

mdevaev, 2021-05-28:
@QEMU90 The new key servers are working. Maybe you have some problem with the network in docker?

wnark, 2021-05-29:
> @QEMU90 The new key servers are working. Maybe you have some problem with the network in docker?

[image]
Hello, I used a server installation in the United States and I also saw this error

mdevaev, 2021-05-29:
It's fine

mdevaev, 2021-05-29:
Fixed image bug: #325. Rebuild your os.

mdevaev, 2021-05-31:
Fixed I guess

closed 2021-05-31 by mdevaev

na784, 2021-06-02:
I had the same problem, traced it down to Docker was confused by protocol identifier hkps.  When I added :443 to the keyserver URLs it started working perfectly.

mdevaev, 2021-06-02:
Thanks, added this to the code.