pikvm/pikvm #321: Issue with progress bar of file uploading

al-pankov, 2021-05-25:
**Describe the bug**
Progress bar not match file uploading process.

**To Reproduce**
Steps to reproduce the behavior, like:
1. Go to 'KVM-mode'
2. Click on 'Drive - Select image to upload - Upload'
3. See error

**Expected behavior**
PiKVM is connected via WiFi. I tried to upload a 5GB ISO file. The file upload bar filled up after 5 seconds, but the download process continued.

**Screenshots**
[image]

**Desktop (please complete the following information):**
 - OS: Win10
 - Browser: Google Chrome
 - Version: 90
 - VNC client: not used

**Pi-KVM info:**
  - Raspberry Pi board version: RPi 4
  - Pi-KVM platform: v2-hdmi
  - Video capture type: CSI bridge
  - KVMD version: 2.72-1
  - uStreamer version: 3.27-1
  - Linux kernel: pikvm 5.10.23-2-ARCH #1 SMP Thu Mar 18 18:33:35 MSK 2021 armv7l GNU/Linux

mdevaev, 2021-05-26:
Is this problem reproducible or only occurred once?

al-pankov, 2021-05-27:
It's occurs every time with big files, but if file is smaller then progress bar filling looks like normal.
I recorded a screencast that show the problem.
https://user-images.githubusercontent.com/48507823/119850984-b44c0a00-bf16-11eb-8bdd-e8e68518e5e9.mp4

PS. With latest kvmd 2.80 the problem persist.

mdevaev, 2021-06-04:
Does this sound like your problem from the description? https://stackoverflow.com/questions/42516195/xmlhttprequest-wrong-progress

mdevaev, 2021-06-07:
I can't reproduce it on my win10 machine and 6Gb image. Are there any additional conditions?

mdevaev, 2021-06-08:
I've made a server-side download percentage calculation. I think it will solve the problem. It will be available in the next release.

closed 2021-06-08 by mdevaev