pikvm/pikvm #322: Large windows ISOs not booting

sharicov, 2021-05-27:
Large windows ISOs (>2.2GB) not bootable (worked in previous versions of pkvm).
Latest pkvm version (updated today).
Getting error 0xc000000d on boot.
Tried with several WinPE builds - same issue.
Small WinPE works fine (260MB) - same PE build as large ISO.
Mounting via web pikvm interface - can see correctly on host PC, but no boot.
Same large ISOs boot correctly with aioboot and ventoy.
Copied ISOs directly to microSD card, created .completed in /meta.
Fixed permissions from pikvm:
```
kvmd-helper-otgmsd-remount rw
chown kvmd:kvmd /var/lib/kvmd/msd/images/*
chown kvmd:kvmd /var/lib/kvmd/msd/meta/*
kvmd-helper-otgmsd-remount ro
```

Please assist.
Thanks!

mdevaev, 2021-05-27:
Try this: https://github.com/pikvm/pikvm/blob/master/pages/cookbook.md#create-a-microsoft-windows-based-flash-disk-image

sharicov, 2021-05-27:
Thanks for quick response, tried several Linux ISOs:
Ubuntu-budgie-21.04 not booting, not even showing up in the boot menue.
Tiny Core Linux x64 (12) boots normally.
Very strange behavior
Can it be related to the fact that pikvm truncates files larger than 2.2GB?

mdevaev, 2021-05-27:
Yes

sharicov, 2021-05-27:
Both linux and windows?
So how do you handle it?

mdevaev, 2021-05-29:
Large images cannot be booted when emulating a CDROM. It doesn't matter Linux or Windows.

sharicov, 2021-05-29:
Tried uploading iso images via web interface - still didn't work, but I certainly remember booting large isos with pikvm (strelec iso).
How come?

mdevaev, 2021-05-30:
So uploading or booting doesn't work?

sharicov, 2021-05-30:
No, tried both Linux (ubuntu) and WinPE (strelec).
Windows gives 0xc000000d error, Linux doesn't show up in the bios boot menu at all. (same as copying to microSD directly).
That drives the whole ISO boot mechanism useless since you don't have modern bootable ISOs under 2.2 GB.
Please check for regression in the newest versions of pikvm.
Thanks!

mdevaev, 2021-05-30:
Nothing has changed in this place. ISO not when not working with images over 2.2 GB. Big images require flash drive emulation

sharicov, 2021-05-30:
Ok, it is what it is then.
Can you please add flash image creation in the web interface and enable option for read write the image partition?
So one can create, map and write to flash drive emulation without leaving the web interface?
Would be a killer feature!
Thanks

mdevaev, 2021-05-30:
I'd rather do DVD support in the future

sharicov, 2021-05-30:
Rather one would be great!

Zulgrib, 2021-06-23:
DVD emulation would probably solve this issue, most ISO are DVD sized these days for system with fancy GUI.

arch1mede, 2021-06-26:
Yes but the correct DVD emulation is not, this is on the todo list but no ETA as other things are taking more priority.

petrkr, 2021-07-22:
Actually I solved that 2 years ago... that size is hard-coded in OTG mass storage.. Do not know if you will increase this limit (or even disable that check) if that simply start to work.. That time I did not test recompile my own kernel module with such "fix"

https://github.com/torvalds/linux/blob/master/drivers/usb/gadget/function/storage_common.c#L247

mdevaev, 2021-07-24:
It's not so easy:
* https://lkml.org/lkml/2015/3/7/388
* https://linuxehacking.blogspot.com/2013/07/how-to-emulatore-dvd-rom-hardware-usb.html

mdevaev, 2021-08-15:
Fresh related and not accepted patches:
- https://www.spinics.net/lists/linux-usb/msg214025.html
- https://www.spinics.net/lists/linux-usb/msg214026.html
- https://www.spinics.net/lists/linux-usb/msg214027.html (looks wrong)
- https://www.spinics.net/lists/linux-usb/msg214028.html
- https://www.spinics.net/lists/linux-usb/msg214030.html
- https://www.spinics.net/lists/linux-usb/msg214029.html

TheLinuxGuy, 2021-12-26:
Any idea if DVD emulation will be on roadmap anytime in 2022?

mdevaev, 2021-12-26:
@TheLinuxGuy I don't know, sorry. Maybe I'll do it myself or try to find someone who will do it.

Paxy, 2022-03-06:
Workaround:
- Use Rufus or similar ISO to flash conversion tool to write ISO to USB flash
- Use DD tool to create USB flash image
- Upload DD image and use it as Flash mount in PiKVM

TheLinuxGuy, 2023-03-24:
Just wanted to swing by with my annual anniversary comment asking if this feature could be prioritized in order to improve UX and convenience to all pikvm users.

Most common major linux distros which offer livecd booting bundle it on their DVD image. Example: all manjaro downloads are greater than 3GB single .iso images: https://manjaro.org/download/ -- flash mode doesn't work for them.

All workarounds are inconvenient and require many steps to make it work. If there was maybe a batch script "tool" provided by pikvm that does the automatic conversion to workaround this bug/FR then I would be happy with that if it was available and pivkm took care of it mostly for me.

folfy, 2023-10-23:
@mdevaev RFC - How about shipping a pre-build image of a ventoy USB (which is about 22M compressed) with PiKVM as a 95% workaround feature? It would mean that we only would have to script putting the iso/image onto the data partition of the USB image as well. For most systems and cases Ventoy works pretty much out of the box, IMHO this would be the next-best solution besides a real DVD emulation. It would be way easier/faster, than dealing the conversion on your PC all the time, and you'd be able to download the iso's directly from the source via your PiKVM (avoiding having to down- and upload it on your local host).

As a MVP, this would be a simple "convert" button, that would:
- Set MSD to RW and unzip the pre-build ventoy.img.gz to e.g. <iso/img-name>_ventoy.img
- Expand the new ventoy-image (sparse-resize), and the data partition to the required size (with parted), to fit the iso/img to be "converted"
- Mount the data partition via loopback, copy original iso/img to it
- Unmount and set MSD to RO
Since step 3 is limited by the speed of your storage device, ideally we would also support direct download of an image "into" a new ventoy-image. One might consider to also just have one ventoy image, and load all images into there (since Ventoy is multi-boot out of the box).

I'd happily support the implementation of such feature.

mdevaev, 2023-10-25:
@folfy I'm not sure. It looks like a crutch instead of fixing a DVD in the kernel.

folfy, 2023-10-25:
@mdevaev I get that, but I doubt fixing DVD device emulation is gonna be an "easy" task to solve in the near future?
Actually, if the solution didn't get ya sold, what if I could implement it in a way that you could just on-demand mount a Ventoy-image with the *existing* MSD partiton as one virtual storage, allowing you to select and boot any image via Ventoy? It actually might be viable too, and this overall would be useful even *if* we would have a working DVD emulation, I mean Ventoy has more features.

mdevaev, 2023-10-28:
Any solution made now will have to be supported, so I wouldn't want to do any special magic. in fact we'll probably make a dvd next year.

SagnikSRHUSE, 2024-03-10:
Is it possible to know when this would be a thing? Perhaps a rough estimate?

mdevaev, 2024-03-11:
I'm not working on it right now because there are higher priorities. You can use Ventoy as a workaround.

drashna, 2024-03-18:
Sorry to dogpile, but ran into this and was a bit disappointed. 

However, creating an img file, mounting that as writeable, and using the Windows Media creation tool on the resulting USB flash drive image boots for me, in uefi mode without CSM enabled.  Don't even need to use ventoy.

Zitt, 2024-05-22:
Count me as disappointed. 
I need to be able to install DVD images to install OSes like ubuntu.

mdevaev, 2024-05-23:
You can use Ventoy to boot DVD images.

petrkr, 2024-05-23:
Ventoy is paid for PXE.

for images like ubuntu you can emulate DISC instead DVD, because that ISO is hybrid, so it should work same as USB flash.

mdevaev, 2025-01-24:
**PiKVM now supports DVD images**

Starting with KVMD 4.49, **the old 2.2 Gb CD limit was removed**.

PiKVM just learned how to support large DVD images natively, which means that **you no longer need Ventoy to prepare Flash images for PiKVM**. You can even use the official Windows ISO to install the OS! Small ISO images will be handled as CD as before, but for big images PiKVM will switch emulation to DVD mode automatically.

* **How to update:** `pikvm-update`, as usual.

*PS: If you've been dreaming of getting DVD emulation, then your wish has been fulfilled :) So if PiKVM has proved useful to you, you can support our full-time job at PiKVM by purchasing [our hardware](https://pikvm.org/buy) or making a small donation to [Patreon](https://www.patreon.com/pikvm) or [Paypal](http://paypal.com/pikvm).*

[image]

closed 2025-01-24 by mdevaev

ivucica, 2025-06-11:
@mdevaev Out of curiosity, what was the change done to enable DVD as opposed to CD emulation? Quickly looking around, I see a mention of `lun.0` in the codebase which usually suggest something to do with SCSI/iSCSI; what was the change required here?