pikvm/pikvm #324: Support "reduced motion"

abh, 2021-05-29:
The "your mouse pointer is hidden" banner animation makes me nauseous. I tried figuring out where it comes from to help with a patch, but I couldn't find it in any of the pikvm code.

Please support "reduced motion" if the user has requested that: https://webkit.org/blog/7551/responsive-design-for-motion/

For example for this banner, just have it be an overlay that doesn't move the rest of the screen; or if it has to be put on top, then put it on top but don't animate moving the rest of the screen.

abh, 2021-05-29:
[image]

Highlighted it with a purple box. The banner is fine, but the animation that makes it fade up and down (and moves the content) makes me feel sick (from a vestibular disorder).

mdevaev, 2021-05-29:
This banner is generated by the browser itself, I can't change it.

abh, 2021-05-29:
> On May 28, 2021, at 23:20, Maxim Devaev ***@***.***> wrote:
> 
> This banner is generated by the browser itself, I can't change it.
> 
What's the API that makes the banner show? I can make a bug for that instead.

mdevaev, 2021-06-03:
Is it Safari?

mdevaev, 2021-06-18:
Sup?

abh, 2021-06-18:
Yes, Safari!

mdevaev, 2021-06-18:
I'll dust off the macbook and check it out

mdevaev, 2021-06-20:
This API is `requestPointerLock()`. Each browser handles this in its own way and I really can't change this behavior. I think you should either use Chrome or ask Apple to fix it. I'm sorry, but it's impossible to do without this API for relative mouse mode.

abh, 2021-06-21:
Brilliant — thank you. I will make an appropriate bit of feedback for that API.

mdevaev, 2021-06-21:
Perfect

closed 2021-06-21 by mdevaev