pikvm/pikvm #326: Flash SPI firmware to Pro Micro failed

cospotato, 2021-05-29:
**Describe the bug**
Flash SPI firmware to Pro Micro failed

**To Reproduce**
Steps to reproduce the behavior, like:
According to https://github.com/pikvm/pikvm/blob/master/pages/flashing_hid.md#spi-firmware

**Expected behavior**
Flash firmware success

**Screenshots**
```
[root@pikvm hid]# make install
bash -ex -c " \
	current=`cat .current`; \
	if [ 'spi' == 'spi' ] || [ 'spi' == 'aum' ]; then \
		gpio -g write 25 1; \
		gpio -g write 25 0; \
	fi \
"
+ current=spi
+ '[' spi == spi ']'
+ gpio -g write 25 1
+ gpio -g write 25 0
platformio run --environment 'spi' --target upload
Processing spi (platform: atmelavr; board: micro; framework: arduino)
--------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/atmelavr/micro.html
PLATFORM: Atmel AVR (3.3.0) > Arduino Micro
HARDWARE: ATMEGA32U4 16MHz, 2.50KB RAM, 28KB Flash
DEBUG: Current (simavr) On-board (simavr)
PACKAGES:
 - framework-arduino-avr 5.1.0
 - tool-avrdude 1.60300.200527 (6.3.0)
 - toolchain-atmelavr 1.70300.191015 (7.3.0)
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 8 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <HID-Project> 2.6.1
|   |-- <HID> 1.0
|-- <ps2dev> 0.0.3+sha.a043002
|-- <digitalWriteFast> 1.0.0
|-- <SPI> 1.0
Building in release mode
Checking size .pio/build/spi/firmware.elf
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [=         ]  10.0% (used 257 bytes from 2560 bytes)
Flash: [===       ]  35.0% (used 10032 bytes from 28672 bytes)
Configuring upload protocol...
AVAILABLE: custom
CURRENT: upload_protocol = custom
Uploading .pio/build/spi/firmware.hex

avrdude: error: Unable to open the spidev device /dev/spidev0.0
avrdude done.  Thank you.

*** [upload] Error 1
======================================================================================================== [FAILED] Took 2.36 seconds ========================================================================================================

Environment    Status    Duration
-------------  --------  ------------
spi            FAILED    00:00:02.358
================================================================================================== 1 failed, 0 succeeded in 00:00:02.358 ==================================================================================================
make: *** [Makefile:17: upload] Error 1
[root@pikvm hid]#
```

**Desktop (please complete the following information):**
 - OS: [e.g. iOS]
 - Browser [e.g. chrome, safari]
 - Version [e.g. 22]
 - VNC client (if used)

**Pi-KVM info:**
  - Raspberry Pi board version RPi 4
  - Pi-KVM platform v2-hdmi
  - Video capture type CSI bridge
  - KVMD version: `kvmd 2.57-1`
  - uStreamer version: `ustreamer 3.26-1`
  - Linux kernel: `Linux pikvm 5.10.23-2-ARCH #1 SMP Thu Mar 18 18:33:35 MSK 2021 armv7l GNU/Linux`

**Additional context**
Add any other context about the problem here.

mdevaev, 2021-05-29:
Hi. Add line `dtoverlay=spi0-1cs` to `/boot/config.txt`, reboot and try again.

mdevaev, 2021-05-29:
Added this to the doc.

mdevaev, 2021-05-31:
Sup?

cospotato, 2021-05-31:
> Sup?

Sorry, I bought a wrong 2-chan TTL level shifter. The new 4-chan level shifter was shipping. Maybe arrive tommorow. I will try it out and reply some feedback.

cospotato, 2021-06-01:
@mdevaev Successed! Thanks a lot!!

closed 2021-06-01 by cospotato

mdevaev, 2021-06-01:
Ur welcome