pikvm/pikvm #332: Enforce changing Pi-KVM passwords

jbacker, 2021-06-04:
**Is your feature request related to a problem? Please describe.**

I'd consider this as a insecure network appliance in its current state. Especially as the 'Access to Pi-KVM from the Internet' instructions are right below the default user/password. Instructions to change this are provided, but we should be more stringent.

**Describe the solution you'd like**

Please enforce changing passwords on the first visit, for both root and the web interface user, before the appliance can be used (Bonus points for being able to paste SSH pubkeys and disable password login on SSH).

This could be an initial setup (which could be expanded to include other things, e.g. setting the IP address of the Pi-KVM).

**Additional context**

It would be awful to see Pi-KVM's reachable from the outside world with default credentials still set. People being people, we all know how badly things can end with insecure IoT appliances connected directly to the internet with default credentials set. We need to prevent people from doing this the best we can.

(And yes, the audience for Pi-KVM is a bit different, but still, I'd see this happening sooner or later by accident..)

mdevaev, 2021-06-05:
Sounds reasonable

arch1mede, 2021-06-10:
I would add that this is a end users responsibility to do, if you put this on the inet without taking basic security precautions then you or your network will get compromised. Please keep in mind that this would be a hassle for those that never intend on exposing this to the inet and don't need to be forced to do something like this. This would add 1-2 additional steps.

geerlingguy, 2021-08-05:
I think this would be good to do from a different approach... I was following the 'Flashing OS' guide (https://github.com/pikvm/pikvm/blob/master/pages/flashing_os.md), and didn't notice I had to run `rw` to set the system to read-write prior to changing my root password, so hit `passwd: Authentication token lock busy`. And in the web UI terminal, trying to update admin password, got `OSError: [Errno 30] Read-only file system: '/etc/kvmd/.htpasswd.0919ji65'`.

It would be good (IMO) to have filesystem be rw at least for first boot, and show a warning until someone sets at least the root password, since that would be a pretty big security hole. It would also maybe allow an automatic run of pacman autoupdate on first boot which may help with resolving issues if someone buys or first boots an old revision.

arch1mede, 2021-09-19:
[image]

ztmr, 2021-12-25:
A number of thoughts on this topic...

This is my initial steps (roughly):
- `rw`
- ask for non-root username to be created and create it
- put this user to `wheel`
- enable `wheel` in `sudoers`
- ask for ssh key(s) to authorize
- set password (might not be needed, depending on whether sudoers have `NOPASSWD`)
- disable `PermitRootLogin` in `sshd_config`
- `reboot` (just `ro` does not seem to be enough, this messes up the `systemd` apparently)

I haven't checked on how the Web users are managed yet:
- it would be great to have a way to either bind web user to the local user somehow or have a similar "hardening" process similar to the local users as described above
- the web shell should be running as a non-root user configured in the console/ssh setup

Perhaps something like `pikvm-config`, inspired by `rpi-config`, would help with all of this? Perhaps all of this could be preset using the 'pikvm.txt' on the boot partition so that the image itself can be customized upfront?

If using Tailscale (or any other VPN solution), it might be useful to restrict the web/ssh logins to the `tailscale0` interface only -- either using firewall or just making ssh/web listening on it.

As I am thinking about all of this, it would be interesting to see PiKVM integrated with Teleport ;-)

nvtkaszpir, 2021-12-26:
shower thoughts:
- add hardening guide to the docs
- when someone logs in via ssh then add blinking ansi color text to notify about default password
- as above but as web ui popup?
- system upgrade on first start may not always be desired, it should be possible to disable it, especially when developing (but in that case usually people bake their own images
- integrate cloud init on rpi boot so that it could be  auto reconfigured on first launch https://cloudinit.readthedocs.io/en/latest/topics/datasources/nocloud.html

mdevaev, 2021-12-27:
If you need some kind of custom complex setup when loading, you can write your own scripts and put them to `/boot/pikvm-scripts.d`: https://github.com/pikvm/kvmd/blob/master/scripts/kvmd-bootconfig#L139

NiKiZe, 2021-12-28:
An alternative would be to have a random password be generated on first startup, having a default password at all is bad.
The question of course then is how to access this password. (assuming not everyone having a monitor)

It was mentioned 

> I would add that this is a end users responsibility to do, if you put this on the inet without taking basic security precautions then you or your network will get compromised. Please keep in mind [...]

With IPv6 it is more likely that the device is reachable than what it was with IPv4 where NAT is everywhere.
NAT is not security, devices should be secure by default even if put on the public internet or some other insecure environment.

nvtkaszpir, 2021-12-28:
in case of randomly generated passwords the only way to access rpi would be via ssh with keys, and then you can force user password change by root.
Providing salted password hash on install would be better, that's why I suggested cloud-init  (not sure if it works with Arch, though). In that case it would be possible to run cloudinit with no-cloud file placed in /boot.

arch1mede, 2021-12-28:
> NAT is not security, devices should be secure by default even if put on the public internet or some other insecure environment.

NAT is not full security but it is more secure than being fully exposed. I'll add that this is no different than any other appliance that's put onto a network, DVR, IoT etc of which none are hardened...this still falls under the end user's responsibility as no ones network is the same across the board. I will also add that ALLOT of distro's including, I think Rasbian, does not do this and having this enforced effects a much larger group of ppl than those that "think" these methods will work. I don't feel, and this is ONLY my opinion, that this project should ALSO be responsible in securing whatever you decide to put on the network, because if this happens then there is an underlying added responsibility, well my blah blah was hacked...and you said it was hardened etc...

> With IPv6 it is more likely that the device is reachable than what it was with IPv4 where NAT is everywhere.
IPv6 is as much accessible as IPv4 at this point and routes the same, if the router accepts ipv6, but as Max has stated, IPv6 is still in its infancy so this is basically moot.

arch1mede, 2021-12-28:
> in case of randomly generated passwords the only way to access rpi would be via ssh with keys, and then you can force user password change by root. Providing salted password hash on install would be better, that's why I suggested cloud-init (not sure if it works with Arch, though). In that case it would be possible to run cloudinit with NoCoud type config file placed in /boot.
> 
> Edit: looks like that could work https://wiki.archlinux.org/title/Cloud-init

CloudInit assumes inet connectivity, some are not connected to the inet so this would not be a viable solution.

nvtkaszpir, 2021-12-28:
cloud-init does not assume inet connectivity, you can run it on fully isolated systems - that's why nocloud configs are used in for example qemu-kvm with cdrom or floppy device (or just whatever to read a file from device). In that case the only use would be to run modules which do not require internet, for example user accounts, file permissions...