pikvm/pikvm #333: VGA to HDMI Converter

justincase3, 2021-06-07:
Cant seem to find a decent VGA to HDMI for a typical Dell RX720D server with Debian installed.
Tried 2 : 
QGeeM VGA to HDMI Adapter (amazon) :  BIOS screen displays fine on KVM  , even the first part of boot  fine , then a display resolution change happens before a few outputs into linux prompt.  Here the KVM  has greenish garbled screen .
Unplug KVM VGA and plug back a regular VGA monitor : getting "out of range" .   Only a reboot on the Dell  fixes this 

SIIG VGA to HDMI  (CE-VG0U11-S1) :  This one does exactly the opposite, BIOS Screens grabled and chopped and with green tint on the KVM.  as soon a few lines into Debian boot , where the display resolution changes , the KVM is ok now and displays 
the linux login and I can continue to login in and work like normal. 

Any suggestions or related experiences ?

arch1mede, 2021-06-10:
Please join discord and view the pins, there are several that are in use that work well.

justincase3, 2021-06-12:
Appreciate the feedback. It seems the discord invite is expired : https://discord.com/invite/bpmXfz5
Thanks

arch1mede, 2021-06-12:
I had the dev look at the expire and its still active, please check again. You can also try https://discord.gg/bpmXfz5 that's on the bottom of the page on the main website.

arch1mede, 2021-06-26:
Were you able to join the discord?

closed 2021-07-04 by mdevaev