pikvm/pikvm #341: New H.264 not reliable

GregoryOrciuch, 2021-06-11:
**Describe the bug**
Upgraded today by pacman -Syu, rebbot, then enabled kvmd-janus.
The h264 option appeared in the UI so wanted to try to use it.
At first glance everything working well, image quality way better. It works for 1-5 minutes, then it restart and then it keeps
saying "Error connectiong to Janus Websockets is it up?"

In log I can see this one:
```
Jun 11 09:51:46 pikvm kvmd-janus[394]: kvmd.apps.janus.runner            INFO --- => Creating new handle in session 5920712345424902: 8213275485065025; 0xb1d00df8 0xb1d01260
Jun 11 09:51:46 pikvm kvmd-janus[394]: kvmd.apps.janus.runner            INFO --- => == ustreamer -- Creating session 0xb0801348 ...
Jun 11 09:51:46 pikvm kvmd-janus[394]: kvmd.apps.janus.runner            INFO --- => == ustreamer -- Memsink opened; reading frames ...
Jun 11 09:51:48 pikvm kvmd-janus[394]: kvmd.apps.janus.runner            INFO --- => [8213275485065025] Creating ICE agent (ICE Full mode, controlling)
Jun 11 09:51:49 pikvm kvmd-janus[394]: kvmd.apps.janus.runner            INFO --- => [WARN] [8213275485065025] Waiting for candidates-done callback... (slow gathering, are you us>
Jun 11 09:51:50 pikvm kvmd-janus[394]: kvmd.apps.janus.runner            INFO --- => [8213275485065025] The DTLS handshake has been completed
Jun 11 09:52:16 pikvm kvmd-janus[394]: kvmd.apps.janus.runner            INFO --- Got new _Netcfg(nat_type='Full Cone NAT', src_ip='192.168.1.15', ext_ip='', stun_host='stun.l.go>
Jun 11 09:52:16 pikvm kvmd-janus[394]: kvmd.apps.janus.runner           ERROR --- Empty src_ip or ext_ip; stopping Janus ...
Jun 11 09:52:16 pikvm kvmd-janus[394]: kvmd.apps.janus.runner            INFO --- Stopping Janus ...
Jun 11 09:52:21 pikvm kvmd-janus[394]: kvmd.apps.janus.runner            INFO --- Process killed: pid=849; retcode=-9

```
But see, it was working well for a while. Reboot of rpi4 helps here, but the it does fail again after few mins.

**To Reproduce**
Steps to reproduce the behavior, like:
1. Just upgrade and try h264

**Expected behavior**
Well, I'd like the connection to stay for a while

**Desktop (please complete the following information):**
 - OS: [e.g. iOS]
 - Browser [e.g. chrome, safari]
 - Version [e.g. 22]
 - VNC client (if used)

**Pi-KVM info:**
  - Raspberry Pi board version pi4
  - Video capture type csi brige
kvmd 3.3-1
kvmd-platform-v2-hdmi-rpi4 3.3-1
kvmd-webterm 0.39-1

mdevaev, 2021-06-11:
Let me know if it doesn't work, and I'll deal with this bug again.

GregoryOrciuch, 2021-06-11:
way better, thanks!

wnark, 2021-06-24:
> way better, thanks!

Hello, do I need to build a stun server even if I use it on a LAN? If I need to use pikvm on the internal network, do I need to deploy my stun server to the internal network or the external network?

mdevaev, 2021-06-24:
@wnark use this without editing the config: https://github.com/pikvm/pikvm/blob/master/pages/webrtc.md#custom-janus-config