pikvm/pikvm #342: Balena Etcher does not produce a working SD Card for Pi-ZeroW on M1 Mac

SleepIT, 2021-06-11:
Using Balena Etcher on MacOS (M1 Mac Mini) seems to work fine for the Pi4 v2 image but not for Pi0W v2 image.  After MUCH wrangling (different SD cards, different Pi's, etc), I finally tried flashing the card using Balena Etcher in Windows and it worked fine.  After that, I also tested flashing it using the DD command on the Mac and that works as well.

**To Reproduce**
1. Flash Pi0w v2 image to SD using Balena Etcher on M1 Mac
2. Edit pikvm.txt as necessary
3.  Power the pi, wait 10 minutes
4. Remove and reapply power

**Expected behavior**
Expect Pi to connect to wifi and be available on the network but it's not.  Not sure what's happening because I don't have a mini-HDMI monitor cable (yet).

**Desktop **
 - OS: macOS Big Sur (M1 Mac Mini)

**Pi-KVM info:**
  - Raspberry Pi Zero W
  - Pi-KVM platform - v2-hdmi
  - Video capture type - CSI Bridge

**Additional context**
Balena Etcher works fine for flashing Raspbian on the M1 (flash, edit wpa_supplicant.conf, etc) as well as the PI4 v2 image so I'm not sure what it's doing wrong with Pi-KVM.

arch1mede, 2021-06-14:
I am pretty sure this isn't an issue. Something must be going on as Pi-KVM is NOT an OS. It's just services riding on top of an existing OS that's already available. This behavior is normal if you edit pikvm.txt incorrectly. I would close this issue and re-open once you get a mini-hdmi adapter.

SleepIT, 2021-06-14:
Problem found.  TextEdit on macOS does not work for editing pikvm.txt (from what I've read it was because I was using the clipboard to paste it in (which messes up the end-of-line characters), would probably work fine typing it which is why I didn't have this problem when I flashed my first card for PI 0 and didn't have the problem when I edited the wpa_supplicant for Raspbian).  When I used DD and it worked, I edited the file using vi which explains why that worked.

closed 2021-06-14 by SleepIT