pikvm/pikvm #343: RPI3 make os failed Syntax error: Unterminated quoted string, Error 2 makefile

bryanwills, 2021-06-11:
**Describe the bug**
Followed the directions on [https://github.com/pikvm/pikvm/blob/master/pages/building_os.md](url) to make the OS for Raspberry Pi 3B+ and ran into issues building the OS from the Makefile. Tried to build OS for RPI3 on newly updated Ubuntu 21.04, Kernel 5.11.0-18 production machine.

**To Reproduce**
Steps to reproduce the behavior:
1. used the below config.mk file in the os folder after cloning the repo https://github.com/pikvm/os to my machine.
BOARD = rpi3
PLATFORM = V0-hdmiusb
HOSTNAME = pi3kvm
LOCALE = en_US
ROOT_PASSWD = root
WEBUI_ADMIN_PASSWD = admin
IPMI_ADMIN_PASSWD = admin
CARD = /dev/sdc1
WIFI_ESSID = "SSID"
WIFI_PASSWD = "password"
2. Tried to build on production Ubuntu 21.04 machine, and failed with the following error messages: 
```
	--build-arg WEBUI_ADMIN_PASSWD=admin \
	--build-arg IPMI_ADMIN_PASSWD=admin \
	--build-arg NEW_HTTPS_CERT=b8d6e382-78c8-4b30-810b-61c8c9e51e5e \
 \
	./.build
/bin/sh: 27: Syntax error: Unterminated quoted string
make[1]: *** [Makefile:231: os] Error 2
make[1]: Leaving directory '/home/bryanwi09/os/.pi-builder'
make: *** [Makefile:54: os] Error 2
```
3. Created a new VM on ESXI platform running Ubuntu 21.04 and 20.10, both had the same issue when running the sudo make os command.

**Expected behavior**
The os builds from the sudo make os command and proceed to the next step.

**Screenshots**

[image]

**Desktop (please complete the following information):**
 - OS: Ubuntu 21.04 Kernel 5.11.0-18

**Pi-KVM info:**
  - Raspberry Pi board version 3
  - Pi-KVM platform v0-hdmiusb
  - Video capture type HDMI
  
Add any other context about the problem here.

I was able to get around this issue by modifying the Makefile at ~/os/.pi-builder/ at line 55 to read:
```
_BUILD_DIR = ~/os/.pi-builder/.build
```
Not sure why that worked or even if this is an issue, but the build completed after making that change and I could proceed with the installation.

I am trying to install on a Raspberry Pi 3B+ as I had issues with the Raspberry Pi 4B+ getting an image to appear on the piKVM screen. I am trying to connect the piKVM to a Dell PowerEdge 510 that has a VGA output port by connecting a VGA male to HDMI female adapter [https://www.amazon.com/gp/product/B083K83NNX](url) with a known working HDMI to HDMI cord and connecting the other end to the USB to HDMI capture card. [https://www.amazon.com/gp/product/B08CXWPYQ8/](url) I purchased this one after searching online to see what people were using to connect their devices to from testing. I even used the ports as described in the README.md file as well as the settings and updates to the install on the piKVM once the OS install completed.

Once I am able to test the functionality with a working device, I am going to present the data to my employer who is interested in purchasing equipment for the servers that we have that do not have iDRAC connected to them until we phase them out in the next coming years with newer equipment that has a working iDRAC. And I want to use it at home in my lab environment.

mdevaev, 2021-06-12:
Hello! Try in config.mk: `PLATFORM = v0-hdmiusb` instead of `V0`

mdevaev, 2021-06-12:
Also unqoute wifi password. If it requires quotation marks, it is better to configure this later.

arch1mede, 2021-06-13:
**_I had issues with the Raspberry Pi 4B+ getting an image to appear on the piKVM screen._**

Hi, please join discord and we should be able to assist you with this. rpi4 is more recommended than a rpi3, once you get your rpi3 working you are going to experience the same issues.

mdevaev, 2021-06-22:
Answered I guess. If there are still problems, let me know.

closed 2021-06-22 by mdevaev