pikvm/pikvm #344: Help! With the otg mouse

Friday8229, 2021-06-12:
Hi, Thanks a lot for all the work you have done it's amazing. I need help, I want to control the mouse with the help of some push buttons. So can you please tell me where can I find the file responsible for sending the relative mouse movements. What I want is to use four buttons connected to the gpio pins of the raspberry Pi and move the mouse cursor by one pixel whenever the respective button is pressed, any help is appreciated. Please reply as soon as possible.

mdevaev, 2021-06-15:
Hello. See here:
- https://github.com/pikvm/kvmd/blob/master/kvmd/apps/kvmd/api/hid.py
- https://github.com/pikvm/kvmd/blob/master/web/share/js/kvm/mouse.js

Friday8229, 2021-06-20:
Hey I'm sorry but i don't have any idea on how to proceed. I'm a beginner, can you please tell me which part to keep in the hid.py file. All i want to do is provide any function with a value like (0, 1) and it should move the mouse pointer by one pixel on the y axis.

I think the useful function is this one?

   def __send_mouse_move_event_remapped(self, to_x: int, to_y: int) -> None:

mdevaev, 2021-06-22:
Yes, it is. Be specific about your question, I'm not too good at long explanations of how my code works)

Friday8229, 2021-06-22:
Thanks for your reply. Coming to the point after removing every asynchronous function from hid.py and using this command 

__send_mouse_move_event_remapped(self, 0, 1)

The mouse should move by 1 pixel ? Or I have to do something else ?

mdevaev, 2021-06-22:
> The mouse should move by 1 pixel
yes

Friday8229, 2021-06-22:
The file that i modified is attached below, when i run it. I get this error:  

[root@pikvm api]# python3 hid1.py
Traceback (most recent call last):
  File "/usr/lib/python3.9/site-packages/kvmd/apps/kvmd/api/hid1.py", line 90, in <module>
    hid.send_mouse_move_event(0, 1)
NameError: name 'hid' is not defined

[hid.txt](https://github.com/pikvm/pikvm/files/6696327/hid.txt)

Friday8229, 2021-06-22:
Hey, sorry for being a noob. I think i was doing it the wrong way.Now i can execute the command without any error after changing the to_x and to_y values, function shown below but the mouse is not moving, what am i doing wrong?

def __send_mouse_move_event_remapped(self, to_x: 0, to_y: 1) -> None:
        if self.__mouse_x_range != MouseRange.RANGE:
            to_x = MouseRange.remap(to_x, *self.__mouse_x_range)
        if self.__mouse_y_range != MouseRange.RANGE:
            to_y = MouseRange.remap(to_y, *self.__mouse_y_range)
        self.__hid.send_mouse_move_event(to_x, to_y)

mdevaev, 2021-06-25:
You are not calling a function. What are you even trying to do?

Friday8229, 2021-07-05:
My bad, this is the new error.

[root@pikvm api]# python3 hid1.py
Traceback (most recent call last):
  File "/usr/lib/python3.9/site-packages/kvmd/apps/kvmd/api/hid1.py", line 92, in <module>
    hd_api = HidApi()
TypeError: __init__() missing 5 required positional arguments: 'hid', 'keymap_path', 'ignore_keys', 'mouse_x_range', and 'mouse_y_range'

mdevaev, 2021-07-05:
This module will not work separately from the rest of the code.

Friday8229, 2021-07-05:
I just want the mouse part, Can i remove the rest of the code and just supply the needed parameters ? Will it work?

mdevaev, 2021-07-05:
No. I advise you to leave the idea of editing this code and use something simpler, for example this: https://www.isticktoit.net/?p=1383

closed 2021-07-12 by mdevaev