pikvm/pikvm #345: Installation issues, Step 25/91 fails with "Unknown architecture"

arhi, 2021-06-13:
**Describe the bug**
Following https://github.com/pikvm/pikvm/blob/master/pages/building_os.md I cannot make os

**To Reproduce**

prereq arch linux on x86_64 machine, kernel >= 5.8, docker >= 19.03.13
```
[root@arch os]# uname -a
Linux arch 5.12.10-arch1-1 #1 SMP PREEMPT Thu, 10 Jun 2021 16:34:50 +0000 x86_64 GNU/Linux
[root@arch os]# docker --version
Docker version 20.10.7, build f0df35096d
[root@arch os]# cat /etc/os-release
NAME="Arch Linux"
PRETTY_NAME="Arch Linux"
ID=arch
BUILD_ID=rolling
ANSI_COLOR="38;2;23;147;209"
HOME_URL="https://archlinux.org/"
DOCUMENTATION_URL="https://wiki.archlinux.org/"
SUPPORT_URL="https://bbs.archlinux.org/"
BUG_REPORT_URL="https://bugs.archlinux.org/"
LOGO=archlinux
[root@arch os]#
```
First issue 
```
[user@localhost ~]$ curl -fsSL https://get.docker.com -o get-docker.sh
[user@localhost ~]$ sudo sh get-docker.sh
# Executing docker install script, commit: 7cae5f8b0decc17d6571f9f52eb840fbc13b2737

ERROR: Unsupported distribution 'arch'
```
Not a big deal as ```pacman -Su docker``` installs the docker newer than requirement but might explain what's going further

fetching everything and doing make os

```
Step 25/91 : RUN pacman --noconfirm --ask=4 -Syy        && pacman --needed --noconfirm --ask=4 -S               glibc           pacman  && pacman-db-upgrade    && pacman --noconfirm --ask=4 -Syu    && pacman --needed --noconfirm --ask=4 -S               p11-kit                 archlinux-keyring               ca-certificates                 ca-certificates-mozilla               ca-certificates-utils   && pacman --needed --noconfirm --ask=4 -S               base            base-devel              vim             colordiff             tree            wget            unzip           unrar           htop            nmap            iftop           iotop           strace          lsof         git              jshon           rng-tools               bc              ccache  && case "$ARCH" in         arm)             case "$BOARD" in                 rpi|rpi2|rpi3|zero|zerow) pacman --needed --noconfirm --ask=4 -S linux-raspberrypi linux-raspberrypi-headers;;                 rpi4) pacman --needed --noconfirm --ask=4 -S linux-raspberrypi4 linux-raspberrypi4-headers;;                 *) pacman --needed --noconfirm --ask=4 -S linux-armv7-headers;;             esac;;         aarch64) pacman --needed --noconfirm --ask=4 -S linux-aarch64-headers;;         *) echo "Unknown architecture: $ARCH"; exit 1;;     esac      && rm -rf /var/cache/pacman/pkg/*
 ---> Running in 1c6d45e4f360
:: Synchronizing package databases...
error: failed retrieving file 'core.db' from de3.mirror.archlinuxarm.org : Operation too slow. Less than 1 bytes/sec transferred the last 10 seconds
error: failed to update core (download library error)
error: failed retrieving file 'extra.db' from de3.mirror.archlinuxarm.org : Operation too slow. Less than 1 bytes/sec transferred the last 10 seconds
error: failed to update extra (download library error)
downloading community.db...
error: failed retrieving file 'alarm.db' from de3.mirror.archlinuxarm.org : Operation too slow. Less than 1 bytes/sec transferred the last 10 seconds
error: failed to update alarm (download library error)
downloading aur.db...
error: failed to synchronize all databases
The command '/bin/sh -c pacman --noconfirm --ask=4 -Syy         && pacman --needed --noconfirm --ask=4 -S               glibc           pacman  && pacman-db-upgrade    && pacman --noconfirm --ask=4 -Syu    && pacman --needed --noconfirm --ask=4 -S               p11-kit                 archlinux-keyring               ca-certificates              ca-certificates-mozilla          ca-certificates-utils   && pacman --needed --noconfirm --ask=4 -S               base            base-devel              vim             colordiff             tree            wget            unzip           unrar           htop            nmap            iftop           iotop           strace          lsof         git              jshon           rng-tools               bc              ccache  && case "$ARCH" in         arm)             case "$BOARD" in                 rpi|rpi2|rpi3|zero|zerow) pacman --needed --noconfirm --ask=4 -S linux-raspberrypi linux-raspberrypi-headers;;                 rpi4) pacman --needed --noconfirm --ask=4 -S linux-raspberrypi4 linux-raspberrypi4-headers;;                 *) pacman --needed --noconfirm --ask=4 -S linux-armv7-headers;;             esac;;         aarch64) pacman --needed --noconfirm --ask=4 -S linux-aarch64-headers;;         *) echo "Unknown architecture: $ARCH"; exit 1;;     esac      && rm -rf /var/cache/pacman/pkg/*' returned a non-zero code: 1
make[1]: *** [Makefile:231: os] Error 1
make[1]: Leaving directory '/root/os/.pi-builder'
make: *** [Makefile:54: os] Error 2
[root@arch os]# 
[root@arch os]# cat config.mk
BOARD = rpi3
PLATFORM=v0-hdmi
HOSTNAME = pikvm
LOCALE = en_US
TIMEZONE = Europe/Belgrade
ROOT_PASSWD = xxxxxx
WEBUI_ADMIN_PASSWD = xxxxx
IPMI_ADMIN_PASSWD = xxxxxxx
WIFI_ESSID = "xxx"
WIFI_PASSWD = "xxxxx"

```

**Expected behavior**
```make os ``` to finish without error

**Desktop (please complete the following information):**
 - OS: arch 
 - Browser n/a
 - Version n/a
 - VNC client n/a

**Pi-KVM info:**
  - Raspberry Pi board version: trying to make for RPI3B+
  - Pi-KVM platform v0-hdmi

arhi, 2021-06-13:
I freshly installed arch linux for this purpose, I normally use centos or oracle linux, rhel here and there and if I have to debian... I can setup any distro not a big deal but https://github.com/pikvm/pikvm/blob/master/pages/building_os.md requires arch so I used arch .. I see you use apt-get there, not sure why as pacman is kinda arch default package manager, also docker don't really care about arch linux neither... maybe some other distro would be better to create image ?

arhi, 2021-06-13:
changed config to make for RPI2, same issue

```
Step 24/91 : RUN sed -i -e "s/^CheckSpace/#!!!CheckSpace/g" /etc/pacman.conf
 ---> Running in 878b3c77cbb3
Removing intermediate container 878b3c77cbb3
 ---> c397b43ea106
Step 25/91 : RUN pacman --noconfirm --ask=4 -Syy        && pacman --needed --noconfirm --ask=4 -S               glibc           pacman  && pacman-db-upgrade    && pacman --noconfirm --ask=4 -Syu      && pacman --needed --noconfirm --ask=4 -S           p11-kit                 archlinux-keyring               ca-certificates                 ca-certificates-mozilla                 ca-certificates-utils   && pacman --needed --noconfirm --ask=4 -S          base             base-devel              vim             colordiff               tree            wget            unzip           unrar           htop            nmap            iftop           iotop           strace          lsof                git             jshon           rng-tools               bc              ccache  && case "$ARCH" in         arm)             case "$BOARD" in                 rpi|rpi2|rpi3|zero|zerow) pacman --needed --noconfirm --ask=4 -S linux-raspberrypi linux-raspberrypi-headers;;                 rpi4) pacman --needed --noconfirm --ask=4 -S linux-raspberrypi4 linux-raspberrypi4-headers;;                 *) pacman --needed --noconfirm --ask=4 -S linux-armv7-headers;;             esac;;         aarch64) pacman --needed --noconfirm --ask=4 -S linux-aarch64-headers;;         *) echo "Unknown architecture: $ARCH"; exit 1;;     esac    && rm -rf /var/cache/pacman/pkg/*
 ---> Running in 35e05c9ba83c
:: Synchronizing package databases...
error: failed retrieving file 'core.db' from de3.mirror.archlinuxarm.org : Operation too slow. Less than 1 bytes/sec transferred the last 10 seconds
error: failed to update core (download library error)
error: failed retrieving file 'extra.db' from de3.mirror.archlinuxarm.org : Operation too slow. Less than 1 bytes/sec transferred the last 10 seconds
error: failed to update extra (download library error)
downloading community.db...
error: failed retrieving file 'alarm.db' from de3.mirror.archlinuxarm.org : Operation too slow. Less than 1 bytes/sec transferred the last 10 seconds
error: failed to update alarm (download library error)
downloading aur.db...
error: failed to synchronize all databases
The command '/bin/sh -c pacman --noconfirm --ask=4 -Syy         && pacman --needed --noconfirm --ask=4 -S               glibc           pacman  && pacman-db-upgrade    && pacman --noconfirm --ask=4 -Syu      && pacman --needed --noconfirm --ask=4 -S           p11-kit                 archlinux-keyring               ca-certificates                 ca-certificates-mozilla                 ca-certificates-utils   && pacman --needed --noconfirm --ask=4 -S  base             base-devel              vim             colordiff               tree            wget            unzip           unrar           htop            nmap            iftop           iotop           strace          lsof                git             jshon           rng-tools               bc              ccache  && case "$ARCH" in         arm)             case "$BOARD" in                 rpi|rpi2|rpi3|zero|zerow) pacman --needed --noconfirm --ask=4 -S linux-raspberrypi linux-raspberrypi-headers;;                 rpi4) pacman --needed --noconfirm --ask=4 -S linux-raspberrypi4 linux-raspberrypi4-headers;;                 *) pacman --needed --noconfirm --ask=4 -S linux-armv7-headers;;             esac;;         aarch64) pacman --needed --noconfirm --ask=4 -S linux-aarch64-headers;;         *) echo "Unknown architecture: $ARCH"; exit 1;;     esac    && rm -rf /var/cache/pacman/pkg/*' returned a non-zero code: 1
make[1]: *** [Makefile:231: os] Error 1
make[1]: Leaving directory '/root/os/.pi-builder'
make: *** [Makefile:54: os] Error 2
[root@arch os]#
```

mdevaev, 2021-06-15:
Hi. It looks like you have some network problems with access to `de3.mirror.archlinuxarm.org`. Selecto some another using `config.mk` and option `REPO_URL`. https://archlinuxarm.org/about/mirrors

arhi, 2021-06-15:
> It looks like you have some network problems with access to `de3.mirror.archlinuxarm.org`. 

Thanks, for some reason transparent proxy on my pfsense decided to block access to repos (all other url's work ok ?!?!?), thanks again, def not a bug :)

closed 2021-06-15 by arhi

mdevaev, 2021-06-15:
Welcome!