pikvm/pikvm #346: Arduino HID with a Leonardo Board

Steve-Tech, 2021-06-14:
Hey I was just wondering if it's possible to run the Arduino HID with an Arduino Leonardo board.

I've changed the board in `platformio.ini` to `board = leonardo` and it seems to compile and upload successfully, I connected it using the diagram in [Setting up the v0](https://github.com/pikvm/pikvm/blob/master/README.md#setting-up-the-v0) (using the TTL serial pins), but it doesn't work and it leaves the following errors in the PiKVM log.
```[2021-06-14 22:04:05 kvmd.service] --- kvmd.plugins.hid._mcu            ERROR --- No response from HID: request=b'3\x10\x00\x00\x00\x00\x1b\xc4'
[2021-06-14 22:04:05 kvmd.service] --- kvmd.plugins.hid._mcu            ERROR --- No response from HID: request=b'3\x10\x00\x00\x00\x00\x1b\xc4'
[2021-06-14 22:04:05 kvmd.service] --- kvmd.plugins.hid._mcu            ERROR --- No response from HID: request=b'3\x10\x00\x00\x00\x00\x1b\xc4'
[2021-06-14 22:04:05 kvmd.service] --- kvmd.plugins.hid._mcu            ERROR --- No response from HID: request=b'3\x10\x00\x00\x00\x00\x1b\xc4'
[2021-06-14 22:04:05 kvmd.service] --- kvmd.plugins.hid._mcu            ERROR --- No response from HID: request=b'3\x10\x00\x00\x00\x00\x1b\xc4'
[2021-06-14 22:04:05 kvmd.service] --- kvmd.plugins.hid._mcu            ERROR --- Can't process HID request due many errors: b'3\x10\x00\x00\x00\x00\x1b\xc4'
```

The Reset HID button does reset the Arduino though.

I have used the Arduino before and it works fine, but the Logic Level converter could be faulty but it seems to be working when I tried probing it with a multimeter.

**Pi-KVM info:**
  - Raspberry Pi board version: RPi 3B
  - Pi-KVM platform: v0-hdmi
  - Video capture type: CSI bridge
  - KVMD version: `kvmd 3.5-1`
  - uStreamer version: `ustreamer 4.4-1`
  - Linux kernel: `Linux pikvm 5.10.23-2-ARCH #1 SMP Fri Mar 19 00:22:18 MSK 2021 armv7l GNU/Linux`

mdevaev, 2021-06-15:
Check if you have mixed up the rx and tx

mdevaev, 2021-06-29:
Sup?

Steve-Tech, 2021-06-29:
Sorry I didn't see your first comment.
It didn't work still.

mdevaev, 2021-06-29:
The error that you encountered means that communication with the Arduino is impossible. In most cases, this is a schema error. Please check it carefully and make sure that the connections are reliable.

Steve-Tech, 2021-06-29:
The Arduino shows up as a HID in `lsusb` so it should be running the code.
I'll check the connections again I guess.

Steve-Tech, 2021-06-29:
I ended up getting it working, I think the serial headers on my Pi are broken or something because it works when using another Arduino as a USB to Serial adapter.

closed 2021-06-29 by Steve-Tech