pikvm/pikvm #347: override.yaml list of possible settings

mradeschnig, 2021-06-14:
I would like to set the mjpeg quality and fps settings in the override.yaml.
But how are these named. Is a documentation on this?

I would highly recommend to create such documentations, with possible tags and list of values with a description.

mdevaev, 2021-06-14:
Unfortunately, this takes time, which I don't have at the moment, as I'm busy with the code. However, you can view a list of all parameters using the command `kvmd -m`.

mradeschnig, 2021-06-14:
Thank you, I understand. I will try to figure that out.

closed 2021-06-14 by mradeschnig

mcgarrah, 2024-04-08:
For future folks looking for all options in ```override.yaml``` in latest you can also create files in ```/etc/kvmd/override.d``` to add these options as well as separate files. This was produced April 2024 on a PiKVM V3.

```
# kvmd -m
ipmi:
    auth:
        file: /etc/kvmd/ipmipasswd

    kvmd:
        timeout: 5.0
        unix: /run/kvmd/kvmd.sock

    server:
        host: '::'
        port: 623
        timeout: 10.0

    sol:
        device: ''
        proxy_port: 0
        select_timeout: 0.1
        speed: 115200

janus:
    check:
        interval: 10.0
        retries: 5
        retries_delay: 5.0

    cmd:
        - /usr/bin/janus
        - --disable-colors
        - --plugins-folder=/usr/lib/ustreamer/janus
        - --configs-folder=/etc/kvmd/janus
        - --interface={src_ip}
        - --stun-server={stun_host}:{stun_port}
    cmd_append: []
    cmd_remove: []
    stun:
        host: stun.l.google.com
        port: 19302
        retries: 5
        retries_delay: 5.0
        timeout: 5.0

kvmd:
    atx:
        # type: ''
        type: disabled

    auth:
        enabled: true
        external:
            type: ''

        internal:
            file: /etc/kvmd/htpasswd
            force_users: []
            type: htpasswd

    gpio:
        drivers:
            __gpio__:
                device: /dev/gpiochip0
                type: gpio

        scheme:
            __v3_usb_breaker__:
                busy_delay: 0.2
                driver: __gpio__
[... 336 lines cut ...]
        keepalive:
            count: 3
            enabled: true
            idle: 10
            interval: 3

        max_clients: 10
        no_delay: true
        port: 5900
        tls:
            ciphers: ALL:@SECLEVEL=0
            timeout: 30.0
            x509:
                cert: /etc/kvmd/vnc/ssl/server.crt
                key: /etc/kvmd/vnc/ssl/server.key

    streamer:
        timeout: 5.0
        unix: /run/kvmd/ustreamer.sock

watchdog:
    interval: 30
    rtc: 0
    timeout: 300
```