pikvm/pikvm #348: HowTo Little OLED Screen

yveszemp, 2021-06-15:
Which OLED Screen are Supported, and on which Pins they are connected. Needs a Software Modification or is this always enabled ? 
Regards Yves

mdevaev, 2021-06-17:
You can use any OLED screen compatible with Raspberry Pi. Out of the box, the Pi-KVM does not require any screens.

yveszemp, 2021-06-17:
Thanky for your reply. I see that the Kit that we can Buy had a 4 Pin Header for a OLED Screen. But if i install a OLED Screen to the i2c Bus this dont show anything. Any try to install the Adafruit Library and start a Screen Python script fails. Where can i find the Script for the OLED Screen shown at the Pictures ?

arch1mede, 2021-06-17:
https://github.com/pikvm/pikvm/blob/master/pages/v3.md

srepac, 2021-06-17:
The necessary library is already loaded in Pi-KVM to use OLED screen.  You just have to enable the kvmd-oled service to show information on the OLED.  The default kvmd-oled script is not as helpful as the updated kvmd-oled script I wrote up, which is located at https://pastebin.com/u/srepac      The script in my pastebin also includes instructions on how to make it work with the v2 image.

mdevaev, 2021-06-18:
I believe these are the full answers :)

closed 2021-06-18 by mdevaev

Brandoskey, 2021-08-14:
> The necessary libraries are already loaded in Pi-KVM to use the OLED screen. You just have to enable the kvmd-oled service to show information on the OLED. The default kvmd-oled script is, in my opinion, not as helpful as the updated kvmd-oled script I wrote up, which is located at https://pastebin.com/u/srepac. The script in my pastebin also includes instructions on how to make it work with the v2 image so v2 hardware users can also take advantage of it.

Thanks for this, got the oled script working with the oled display that comes with my pi-rack