pikvm/pikvm #351: Force / Fake HDMI input

grandrew, 2021-06-17:
**Is your feature request related to a problem? Please describe.**
No

**Describe the solution you'd like**
I have a display that I look at visually. It is 1920x1080. I do not want PIKVM to get any video input, just send the mouse events. 

Currently there is no way to set the "fake" input so that I can "fullscreen" it and use my keyboard/mouse on a local chromebook to control the screen with no additional HDMI hadrware while just seeing the screen with my eyes

**Describe alternatives you've considered**
A way to configure the screen size so that it "forces" black/blank screen of desired resolution BOTH for web and VNC. Currently VNC just taps out if there is no video input

mdevaev, 2021-06-18:
1. Create black jpeg file with required geometry and place it to `/etc/kvmd/blank.jpeg`.
2. Edit `/etc/kvmd/override.yaml` and add some options:
```yaml
kvmd:
    streamer:
        cmd_append: ["--blank=/etc/kvmd/blank.jpeg"]
```
3. `systemctl restart kvmd` and don't plug any video capture devices.

mdevaev, 2021-06-22:
Let me know if you need anything else.

closed 2021-06-22 by mdevaev