pikvm/pikvm #352: Stuck keys when using shift key

codisms, 2021-06-19:
**Describe the bug**
When connected to a Windows laptop via VNC, keys are sometimes stuck while typing, but always when used with the shift key. For example, entering the key "?" will result in a trail of slash characters ("/") until another key is pressed. (e.g. "?///////////////////////")

**To Reproduce**
Steps to reproduce the behavior, like:
1. Connect to Windows laptop via VNC.
2. Open Notepad, command prompt, Cygwin, PowerShell, etc.
3. Begin typing with use of characters requiring the shift key. (e.g., "?", "<", ">", "|", "{", "}", double quote...)
4. See trailing characters

**Expected behavior**
When a key is pressed, it shouldn't stick the key.

**Desktop (please complete the following information):**
 - OS: Windows 10
 - VNC client: Jump Desktop

**Pi-KVM info:**
  - Raspberry Pi board version RPi 4a
  - Pi-KVM platform: v2-hdmi
  - Video capture type: CSI bridge
  - KVMD version: 3.6.1
  - uStreamer: version: 4.4.1
  - Linux kernel: Linux pikvm 5.10.23-2-ARCH #1 SMP Thu Mar 18 18:33:35 MSK 2021 armv7l GNU/Linux

Note:
I don't know if this is related, but I also get random presses of the escape key.

mdevaev, 2021-06-19:
Could you try TigerVNC? What you described is similar to the client's problems.

codisms, 2021-06-20:
I downloaded TigerVNC and tried it. In my quick testing, the shift key doesn't seem to stick the key, but the random escape key still occurs. I also have an issue where if I copy something from my Mac and then Cmd-Tab into the VNC session, the contents of the clipboard are automatically pasted. Should I open separate issues for these?

codisms, 2021-06-20:
I connected using TigerVNC and ran `showkey` in a WSL window. It reports that it periodically receives a null value as a key press.

mdevaev, 2021-06-20:
What is "null value"?

codisms, 2021-06-21:
Key code zero.

codisms, 2021-06-21:
[image]

codisms, 2021-06-21:
This is an issue with the machine, not the KVM.

closed 2021-06-21 by codisms

mdevaev, 2021-06-22:
I love problems that fix themselves :) Let me know if there are any other bugs with this.