pikvm/pikvm #353: Confusing about back powering issue with Pi zero w

SachiaLanlus, 2021-06-20:
As the conclusion in both the #225 and #116 ,
that 2 power source at the same time is definitely not a good idea.
However, [Novaspirit Tech](https://www.youtube.com/watch?v=xCL5AS0t5Is&t=69s) mentioned that because pi zero is a underpowered device, it is able to be powered through the same port that plugged into the pc.
I just wondering that if I would like to use pi zero w to build the pi-kvm,
can I only plug 1 port to pc and left another empty?
Except for under power issue, is there any issue I should notice or worry about in this scenario?
Thank you

wnark, 2021-06-24:
If your computer is a regular brand, the current of USB2.0 is sufficient for Raspberry Pi Zero. It has been measured with a USB ammeter, high load only needs 100-200ma current

mdevaev, 2021-06-24:
This ^

SachiaLanlus, 2021-06-25:
In my test, the maximum total system power draw is 410mA.
At that time, the cpu utilization is 90~100%.
And it works well now.
Thank you

But I just notice another issue, that sometime the HID will not work.
And just a reboot can fix it.
So when I reboot the pikvm, I should check whether HID work or not.
It just a minor issue, so it is fine for me.

closed 2021-06-25 by SachiaLanlus

wnark, 2021-06-25:
> In my test, the maximum total system power draw is 410mA.
> At that time, the cpu utilization is 90~100%.
> And it works well now.
> Thank you
>
> But I just notice another issue, that sometime the HID will [...]

[image]
It may be related to the power consumption of the external device used，Not much difference haha

SachiaLanlus, 2021-06-25:
> [image]
> It may be related to the power consumption of the external device used，Not much difference haha

Thank you for sharing your experience!
I try to measure again, and the average power consumption is just like yours.