pikvm/pikvm #358: Offer PiKVM as a package

new-sashok724, 2021-06-24:
I would like to have ability to install PiKVM as a package for ArchLinuxARM, it would be much more convenient than running in docker or migrating to PiKVM OS. I think it can be done relatively easily, if PiKVM OS is based on Arch. It also may be first step to offer packages for other OSes in future.

mdevaev, 2021-06-26:
Hi. Pi-KVM is **already** a set of packages for Arch Linux ARM. An entire OS image is offered to simplify configuration. If you are interested in the sequence of necessary commands, then you need to look at the [os](https://github.com/pikvm/os) repository.

closed 2021-06-26 by mdevaev

new-sashok724, 2021-06-26:
Okay, thanks! Then maybe it could be better documented, because i did not find any mention of a repository for ArchLinuxARM with packages or installation instruction for ArchLinuxARM.

mdevaev, 2021-06-26:
Because these are unofficial packages in our own repo.