pikvm/pikvm #359: Is webrtc only available for Raspberry Pi 4b?

wnark, 2021-06-24:
[image]
```
[root@ipmi ~]# systemctl status kvmd-janus
* kvmd-janus.service - Pi-KVM - Janus WebRTC Gateway
     Loaded: loaded (/usr/lib/systemd/system/kvmd-janus.service; enabled; vendor preset: disabled)
     Active: active (running) since Wed 2021-05-19 22:11:58 CST; 1 months 5 days ago
   Main PID: 305 (kvmd-janus)
      Tasks: 11 (limit: 1750)
        CPU: 3.417s
     CGroup: /system.slice/kvmd-janus.service
             |-305 /usr/sbin/python /usr/bin/kvmd-janus --run
             `-322 /usr/bin/janus --disable-colors --plugins-folder=/usr/lib/ustreamer/janus --configs-folder=/etc/kvmd/janus --interface=192.168.31.66 --stun-server=stun.l.google.com:19302

Jun 24 17:07:32 ipmi kvmd-janus[305]: kvmd.apps.janus.runner            INFO --- => [ERR] [src/plugin.c:_clients_thread:224] [ustreamer] Can't open memsink: No such file or directory
Jun 24 17:07:53 ipmi kvmd-janus[305]: kvmd.apps.janus.runner            INFO --- => [WSS-0x70800b60] Destroying WebSocket client
Jun 24 17:07:53 ipmi kvmd-janus[305]: kvmd.apps.janus.runner            INFO --- => Destroying session 8045646714052955; 0x71d00df8
Jun 24 17:07:53 ipmi kvmd-janus[305]: kvmd.apps.janus.runner            INFO --- => Detaching handle from ustreamer; 0x71d01260 0x70801030 0x71d01260 (nil)
Jun 24 17:07:53 ipmi kvmd-janus[305]: kvmd.apps.janus.runner            INFO --- => == ustreamer -- Removing session 0x70801030 ...
Jun 24 17:07:54 ipmi kvmd-janus[305]: kvmd.apps.janus.runner            INFO --- => == ustreamer -- No active watchers, memsink disconnected
Jun 24 17:07:57 ipmi kvmd-janus[305]: kvmd.apps.janus.runner            INFO --- => Creating new session: 5124019920183655; 0x70802068
Jun 24 17:07:57 ipmi kvmd-janus[305]: kvmd.apps.janus.runner            INFO --- => Creating new handle in session 5124019920183655: 7650495398367946; 0x70802068 0x71d04820
Jun 24 17:07:57 ipmi kvmd-janus[305]: kvmd.apps.janus.runner            INFO --- => == ustreamer -- Creating session 0x71d01070 ...
Jun 24 17:07:57 ipmi kvmd-janus[305]: kvmd.apps.janus.runner            INFO --- => [ERR] [src/plugin.c:_clients_thread:224] [ustreamer] Can't open memsink: No such file or directory
```
Hello, I try to use WebRTC on Raspberry Pi 2B, when I execute it manually
systemctl enable --now kvmd-janus
Although there are open options on the web page, WebRTC cannot be used.
Is webrtc only available for Raspberry Pi 4B, is this determined by the hardware?

wnark, 2021-06-24:
By the way, what does memsink do? I can't find it on Google

mdevaev, 2021-06-24:
Add to `/etc/kvmd/override.yaml`:
```yaml
kvmd:
    streamer:
        h264_bitrate:
            default: 5000
        cmd_append:
            - "--h264-sink=kvmd::ustreamer::h264"
            - "--h264-sink-mode=0660"
            - "--h264-bitrate={h264_bitrate}"
            - "--h264-gop={h264_gop}"
```

mdevaev, 2021-06-25:
PS: Memsink is my own shared memory protocol.

wnark, 2021-06-25:
> Add to `/etc/kvmd/override.yaml`:
>
> ```yaml
> kvmd:
>     streamer:
>         h264_bitrate:
>             default: 5000
>         cmd_append:
>             - "--h264-sink=kvmd::ustreamer::h264"
>             - [...]

[image]
Hello, how do you see the real frame rate of the transmission? I checked the information and found that the Raspberry Pi uses 2-channel CSI, which should only support 1080P 25 frames at most. The higher frame rate is now supported because the tc358743 chip is used to compress the video stream into xh2.54 and then transmit it to the Raspberry Pi ? I find the CPU usage of the Raspberry Pi is not high.

wnark, 2021-06-25:
https://www.raspberrypi.org/forums/viewtopic.php?t=120702&start=575
https://fluxcoil.net/wiki/hardwarerelated/raspberry_pi_4_tc358743
[image]

I have checked some information. Even if the video stream is compressed to h264 using tc358743, the video on the Raspberry Pi 2/3/4B is only up to 1080P50 frames? That is, the picture above me is 50 frames? Although I can't detect it.

wnark, 2021-06-25:
By the way, when I play the game with pikvm, the screen is torn. I don't know if it is because the acquisition rate of tc358743 is not high enough, and it needs 1080P60 frames to be better?

wnark, 2021-06-25:
It’s a bit strange. According to the information found, when the Raspberry Pi 4B is used in conjunction with tc358743, even if 2-csi is used, it can reach 1080P50 frames. This seems to be the limit, but the limit of PIKVM is 25 frames, which is different. The reason for encoding?

wnark, 2021-06-25:
[image]
[image]

In view of the weak CPU performance of the Raspberry Pi zerow and the ability to communicate only via wifi, it is recommended that if you use the Raspberry Pi zerow, only use webrtc to transmit images, which will put a lot of pressure on bandwidth. Even 1080P can run smoothly. Of course, you can also add a w5500 to the Raspberry Pi zerow, but after calculating the cost, it is actually similar to 3B. If you add a network port, it is better to buy a 3B.
The SPI communication speed is 10Mb/s. Using webrtc+zero(w)+w5500 may be the lowest cost solution, and you can even watch videos remotely.
I don't know whether the CPU occupies higher using spi or the CPU occupancy of WiFi.
If anyone is interested, you can try it.

arch1mede, 2021-06-26:
RPI 2X is NOT recommended, RPI3B "shouldn't" be recommended, its just not worth the effort to get one of these working but can understand why one would use it. 0w works but is very slow compared to a RPI4. The max you are going to get from this is 1920x1080p50Hz, 25 is the max fps you are going to get from this from a perfect scenario but the way its being utilized, you are not going to see this. What you are looking at is DYNAMIC FPS.

Please review: https://github.com/pikvm/pikvm/blob/master/pages/Community_FAQ.md#why-so-much-latency-between-the-screen-and-target

I would invest in a RPI4 and reuse your 2/3B's for something else, like pihole or something.

mdevaev, 2021-07-04:
@wnark
> By the way, when I play the game with pikvm, the screen is torn. I don't know if it is because the acquisition rate of tc358743 is not high enough, and it needs 1080P60 frames to be better?

I suppose so.

> This seems to be the limit, but the limit of PIKVM is 25 frames, which is different. The reason for encoding?

The reason is the limited bandwidth of CSI. You can't transfer more than 25 fps. RPi4 has only 2 lanes. For a larger frame rate, you need 4.

Due to the CPU weakness, I don't recommend using WebRTC on anything other than Pi 4. Officially adding this functionality for older boards is a low priority task. I really recommend that you buy a Pi 4 with the smallest amount of RAM and forget about all the problems.

mdevaev, 2021-07-04:
However, if you find a way to use WebRTC, it will be great.

closed 2021-07-04 by mdevaev