pikvm/pikvm #360: The H.264 support for pi zero w

SachiaLanlus, 2021-06-25:
**Is your feature request related to a problem? Please describe.**
Since the launch of H.264 for pi 4b, I just can't sleep and eager for the support for pi zero.

**Describe the solution you'd like**
H.264 support for pi zero.

**Describe alternatives you've considered**
H.264 support for pi zero.

**Additional context**
As the statement in the [document](https://github.com/pikvm/pikvm/blob/master/pages/webrtc.md), H.264 encoding can provide significantly less traffic consumption.
That is very beneficial for the device with poor connection such as pi zero which use 2.4G wifi.
And also, use hardware H.264 encoding can indeed reduce the burden on cpu load, it can spare the juicy cpu resource of the pi zero.
From the statement above, I think the pi zero need the H.264 encoding support more than pi 4b.
Do you have the plan to add the H.264 support for pi zero?
Thank you~

mdevaev, 2021-06-25:
Testing showed that zero will burst into flames and the video will slow down a lot. Don't try to get a miracle from zero, it is a very weak single-core chip with a low frequency. So there are no plans to enable webrtc for zero.

closed 2021-06-25 by mdevaev

SachiaLanlus, 2021-06-25:
So sad, but thank you.
(ノへ￣、)

mdevaev, 2021-06-25:
Yea.