pikvm/pikvm #364: Switch between Arduino HiD and OTG over Webinterface possible?

semool, 2021-06-28:
First, great Software. I have build me a v2 Version with an rpi4-4GB and its working great on my 3 three Nuc Systems (a Proxmox Cluster).

But i have another "old" Machine for Testing things and OTG isnt working (only detecting MSD in Bios). After consulting the Pages on github i have build a Arduino HiD for this Machine and with this second USB Cable the Keyboard is working after is switching the HiD Mode in override.yaml.

Great so far, but...Next week my ezcoo Switch is coming an the all three Nuc Machines and the old one are connected to this.When i control the old machine i must switch to Arduino HiD.I have made me a small script which is working for this:

```
#!/bin/bash
if [ "$1" == "hid" ]; then
   MODE="arduino"
 else
   MODE="otg"
fi
rw
ln -sf /etc/kvmd/override.yaml_$MODE /etc/kvmd/override.yaml
ro
systemctl restart kvmd
exit
```
It is possible to add a Button in the Webif to make this switch? Eventually without a kvmd restart?

Edit:
Eventually with an extension to the ezcoo Code? So that when i switch to this port its automaticly changed to the Arduino HiD and back to otg on the Ports where not have set "hid: arduino"?!

```
ch3_button:
                driver: ez
                pin: 3
                mode: output
                switch: false
                hid: arduino
```

mdevaev, 2021-06-28:
Hi! Thanks.

You definitely can't do that. As you yourself point out, this requires a restart. During the restart, all internal mechanisms and data structures are reinitialized. Moreover, your script will only work if the last HID mode was OTG before rebooting, since it requires special settings for the USB controller.

semool, 2021-06-29:
Ok thank you. Then i will wait and buy a v3 when its available for the ezcoo switch and the v2 moves over to the old machine.

mdevaev, 2021-06-29:
Welcome

closed 2021-06-29 by mdevaev