pikvm/pikvm #367: Indicator of server power state

forkicks, 2021-06-29:
**Is your feature request related to a problem? Please describe.**
I could not find a way to tell if the system the Pi-KVM is managing is powered on already or not (for simplicity's sake, assume there is no screen output to look at and tell)

**Describe the solution you'd like**
A status bar (top) indicator of the current "Powered On/Off" state of the managed system

**Describe alternatives you've considered**
I can always try to get output to the hdmi capture port, but if the server is hung/not powered, it is always black and not informative. At that point, I am not certain what the power state is - either powered on but somehow blocked/crashed or powered off.

**Additional context**

mdevaev, 2021-06-29:
Please read the readme carefully. What you need is called an ATX power controller. You need to assemble a small scheme for it.

forkicks, 2021-06-29:
Hi. Thank you for your reply.

I have the atx controller built and working fine. I can power on/power off/reset perfectly. The problem is knowing in which state the server is at any given point, not controlling it.

mdevaev, 2021-06-29:
The ATX controller contains two relays for reading LED signals: power and reset. Connect them and you will see it in the interface.

forkicks, 2021-06-29:
Thank you. Closing this question.

closed 2021-06-29 by forkicks