pikvm/pikvm #368: Cannot view WebRTC stream

krosseyed, 2021-06-30:
**Describe the bug**
I am able to view my computer using the MJPEG stream. However, when I try to view the H.264/WebRTC stream, I get the following output:

[image]

**To Reproduce**
Login and press the H.264/WebRTC button under settings.

**Expected behavior**
An image of the desktop.

**Desktop (please complete the following information):**
 - OS: Windows 10
 - Browser: Firefox 89.0.2
 
**Pi-KVM info:**
  - Raspberry Pi board version: RPi 4
  - Pi-KVM platform: v2-hdmi
  - Video capture type: CSI bridge
  - KVMD version: 3.6.1
  - uStreamer version: 4.4.1
  - Linux kernel: Linux pikvm 5.10.23-2-ARCH #1 SMP Thu Mar 18 18:33:35 MSK 2021 armv7l GNU/Linux

**Additional context**

Log output when connecting to the WebRTC stream:

[2021-06-30 05:09:27 kvmd-janus.service] --- kvmd.apps.janus.runner            INFO --- => [WARN] [6503655205704125] ICE failed for component 1 in stream 1, but let's give it some time... (trickle received, answer received, alert not set)
[2021-06-30 05:09:32 kvmd-janus.service] --- kvmd.apps.janus.runner            INFO --- => [ERR] [ice.c:janus_ice_check_failed:1848] [6503655205704125] ICE failed for component 1 in stream 1...
[2021-06-30 05:09:32 kvmd-janus.service] --- kvmd.apps.janus.runner            INFO --- => [6503655205704125] WebRTC resources freed; 0xb1d0eb28 0xb0802600
[2021-06-30 05:09:32 kvmd-janus.service] --- kvmd.apps.janus.runner            INFO --- => == ustreamer -- Memsink closed
[2021-06-30 05:09:33 kvmd-janus.service] --- kvmd.apps.janus.runner            INFO --- => == ustreamer -- No active watchers, memsink disconnected

Image of my PiKVM setup:

[image]

arch1mede, 2021-06-30:
Please join discord so we can help troubleshoot. In the mean time please try a private/incog window

krosseyed, 2021-06-30:
This issue was somewhat resolved via Discord, and it was determined that there was nothing wrong with the PiKVM. 

My network configuration is causing this issue and I will need to look into why my firewall does not like WebRTC.

closed 2021-06-30 by krosseyed