pikvm/pikvm #371: [Solution] Hardware mod to AIMOS 4 port switch; remote control it and then pass the pikvm in via hub port

SpookHCK, 2021-06-30:
The AIMOS 4 port switch (AM-KVM401K-V2.1 2020-09-19) is based on the AX68004 chip; there is a serial port pin header on the board, but it is unpopulated and seems to be output only...

The problem is that the nvm seems to block/intercept other devices other than the keyboard/mouse.
This could be down to the hotkey switch mode.

I have tried the Arduino helper board but the KVM appears not to recognize this as a valid keyboard/mouse so it also is blocked.

I have worked around this with a little hardware hack.. Now the Keyboard and mouse and Storage all pass through the KVM... "hotkeys" have been replaced with a gui button but you can customize that as below.

Please note the optocoupler board can be either inside the KVM or on the PI if you have other GPIO pins you are controlling with it and want all the circuitry to be near the pi.

Solder on to the back of the switch button.   - 
The pin header is a read only serial port the pins are labeled on the bottom of the pcb.
[image]

Wire the switch through an optocouple.  
[image]

Don’t forget to insulate the bottom.
[image]

Finished nvm hack:
[image]

Parts:
Wire: https://www.amazon.com/gp/product/B01LWI20M0/
Port: https://www.amazon.com/gp/product/B01ASF0N3A/
jack: https://www.amazon.com/gp/product/B07ZT6HGYS/
Optocouple: https://www.amazon.com/gp/product/B00DKYPVUC/
Resistor 390 Ohm - I have some laying around...
Strip board: https://www.amazon.com/gp/product/B00C9NXP94/ 
Pin headers - I have some laying around...  0.1 inch pitch to wire the ground/GPIO on the jack

Look here for detail on customizing the pikvm gpio control:
https://github.com/pikvm/pikvm/blob/master/pages/gpio.md

mdevaev, 2021-06-30:
Perfecto. This is listed in the documentation multiport.md. Really great job, thanks for fharing!

closed 2021-06-30 by mdevaev

SpookHCK, 2021-08-28:
> Just for clarification: With this mod I can switch the AIMOS from webif and use MSD without a second RPi Zero?

Yes, with this mod you can configure a GPIO pin on the PI to control the KVM "next" screen button and then plug the pikvm into the Hub port and you will pass keyboard mouse and MSD to your target without another device. you will not be able to use the [crtl] [ctrl] [ _number_ ] to pick a screen as that only works on the keyboard/mouse port, but that is also the thing that blocks the MSD.

organizedkaos, 2023-01-31:
@SpookHCK 

I know this topic is closed and old but I just recently acquired the aimos 4 port kvm and I am unable to get keyboard/mouse functionality.

I was wondering if you had the wiring schematic for the optocoupler and resistor?  

Also, I found this video here https://www.youtube.com/watch?v=eXzK3U6huwY and was wondering if I could do the same as well as I have resistors and transistors on-hand but not the optocoupler.

alexisfrjp, 2024-07-11:
That's exactly what I was thinking doing since all the switches with hotkey are quite expensive.

Majestic7979, 2024-12-07:
I have a similar KVM that only works with two devices, so the switch button only alternates between PC1 and PC2. In that case, I was wondering if I could just use the gpio config like this:
```
kvmd:
    gpio:
        scheme:
            relay2:  # Channel 2
                driver: relay
                pin: 1
                mode: output
                initial: null
                pulse:
                    delay: 2  # Default pulse value
                    max_delay: 2  # The pulse interval can be between min_delay=0.1 (by default) and max_delay=2
```
Then of course I'd add the webui controls, etc. And when I tap the button on the WebUI it would send a "press" via the GPIO GND+PIN, closing the circuit and making the KVM switch to the other channel?

I don't think anything else is necessary, right?

Why are you using a mosfet and a resistor, could you educate me why please? Is it because it's 4 ports so you need more components? Mine is just PC1 and PC2 so the button to select is merely a momentary switch.