pikvm/pikvm #374: Running pikvm on top of another OS

dustymabe, 2021-07-04:
Thanks for giving the world pikvm! I'm looking forward to possibly picking up one of these once the v3 is available for purchase.

I'm wondering... If I wanted to run pikvm on top of another OS (say Fedora or Ubuntu) how much effort would be involved there. I'm an OS hacker of sorts so I should be able to handle relatively advanced configuration/troubleshooting. Just figured I'd come to the experts and ask before I started on that journey in the future. I assume the answer will either be one of: "shouldn't be too hard" or "it would be really really hard". Just want to know which camp I'm in.

Thanks!

arch1mede, 2021-07-06:
I look forward to this port, its a daunting task though and many have failed to port it. You wont need the v3 to do this, all v2 hardware will work

mdevaev, 2021-07-16:
Related: #245

srepac, 2021-09-12:
I have created a PiKVM installer for Raspbian OS.  It's one script that can be run on either Raspbian Lite or Desktop 32-bit OS.

mdevaev, 2021-09-12:
Please note this is unofficial and don't work with debian package manager
and may cause problems with update.

srepac, 2021-09-12:
Raspbian works to install .deb packages via dpkg and I have also performed updates of pikvm "packages" via an update script.

srepac, 2021-09-12:
https://kvmnerds.com/RPiKVM/How_to_Install_PiKVM_onRaspbian_32-bit_OS.pdf
https://kvmnerds.com/RPiKVM/install-pikvm-raspbian.sh

See for yourself.  The installer actually performs dpkg -i <*.deb> packages for some of the pre-requisite packages for ustreamer and ttyd (webterm)

mdevaev, 2021-09-12:
This is a smallest part of distribution.

mdevaev, 2021-10-17:
I suppose it can still be considered a discussion and not a specific task. There is no movement here, all issues have been resolved, so I'm closing it.

closed 2021-10-17 by mdevaev