pikvm/pikvm #375: Keyboard Code Problem?

semool, 2021-07-04:
**Describe the bug**
When i in Webif and will Type | or @ at my german Keyboard on a german Windows that dont work.
When i go over Shortcuts and Paste and type in the field | he writes a ' after i press the paste button, and by @ he writes a "

When i connect with VNC i can type without Problems.

PIKVM Version: 3.7

mdevaev, 2021-07-04:
Make sure that you have selected the correct keyboard layout on the managed computer. It must correspond to the local one. Pi-KVM cannot switch the layout itself.

semool, 2021-07-05:
Yes i have check it and i have set it with loadkeys de.

semool, 2021-07-05:
OK deeper investigate.
When i open the PiKVM onscreen Keyboard i can press the right Alt Button and then the Q and he writes a @.
When i press on my physical Keyboard the right Alt Key ("ALT GR" on german Keyboard) the onscreen Keyboard pressed right alt + CTRL ... but when i then press Q nothing happens.
And i cant write a pipe ("|") with the onscreen, the Button dont exist on the onscreen Keyboard. Its the <> Button right after the left shift key on german Keyboard.

semool, 2021-07-05:
OK after some research i can type the pipe with "ALT + NmLck 1 + 2 + 4.
But i think somewhere in the PiKVM Keymap is a bug or missing Key that it dont work directly?!

semool, 2021-07-05:
Thats drives me crazy :) Another way to get the pipe is on the onscreen Keyboard: Alt Right + N/US Key.
OK i dont understand all this Keyboard Stuff but this is enough for me.
The main thing is that I can enter the characters :)

mdevaev, 2021-07-06:
The fact is that I also do not fully understand the logic. I added non-us at the request of another person with a German keyboard and he said that everything works fine. If you understand how this should be used, could you add a detailed description of the work? Sorry :)

semool, 2021-07-07:
In fact its realy simple.
On the german Keyboard we have a unique Button: https://de.wikipedia.org/wiki/Tastaturbelegung#Besonderheiten_der_QWERTZ-Variante
The one with <> and |

This Button is the N/US Key. It is OK.

When we type a | or € or @ on the german Keyboard we are type "Alt Gr" and the corresponding Button:
€ = E, | = <>| (N/US), @ = Q

All this is working with the Onscreen Keyboard in Webif BUT with the physical Keyboard the "Alt Gr" Button dont work.
I think the Problem is that when i press "Alt Gr" on the physical Keyboard on the onscreen Keyboard the Keys "Alt right" AND "CTRL" gets activated.
And that is the Problem, "CTRL" must not be operated when i press "Alt Gr" on phisycal Keyboard.

PfiatDe, 2021-07-22:
I am facing the same problem and a solution to this would be great. 
German Keyboard Layout on Server and on PIKVM.

mdevaev, 2021-07-24:
It's hard for me to deal with this problem because I don't have a keyboard with the right layout. I will try to find the time or delegate it to someone.

SchoutenAtConet, 2021-08-17:
Can you please describe where (in which file) I can change the current mapping for ALTGR to ALT_RIGHT? I wasn't able to spot it.
I would like to make that change and test it.

SchoutenAtConet, 2021-08-17:
I tested using the DE keyboard in the Webif on linux.
The problem is ONLY if I use win 10. On debian 10 the keyboard is totally fine.

RC-Thoughts, 2021-10-03:
I can confirm the same behavior with FI keyboard, "loadkeys fi" tried.

marcelser, 2021-12-14:
Hi Sorry to chime in but on Swiss-German it's even worse and it doesn't even work in VNC alghough I have selected "de-ch" for Keymap value in /etc/kvmd/override.yaml.

I have confirmed that when I locally connect Keyboard through USB it works perfectly including the special keys that need the "Alt-Gr" key in combination. In WebUI and VNC all the "Alt Gr"+Key combinations don't work and on Swiss German Layout and there are very important characters bound to "Alt Gr"+Key combinations like [ ] { } | \ # @

You can imagine if those characters can not be used not even in VNC it renders PiKVM Keyboard emulation pretty usesless. It makes it more or less impossible to operate a linux machine without those keys especially as there's no copy paste functionality.

I have three interesting things though. 
1.) In "virtual keyboard" the correct keys are highlighted when I press "Alt Gr"-ey which is translated to "Alt+Ctrl" beeing pressed. This also works locally I can use "ALT+CTRL"+Key (instead of "Alt Gr"+Key) also to produce the same characters so "Alt Gr" Key is more or less just a shortcut for "ALT+CTRL" together. But in PiKVM the key combination "ALT+CTRL" also doesn't work, it doesn't produce any output
2.) If I copy those non working keys from a notepad into clipboard and try to paste them in the vnc client the copied characters are converted to something else. So If I copy `[ ] { } | \ # @` and paste it into VNC client it turns into `ü "è ! £ $ * "` 
The really interesting thing about this is. The characters they turn into is their "Shift"+Key counterpart. So it's like instead "ALT+CTRL"+Key is pressed, "Shift"+Key is pressed on the USB side and that's probably where the bug is.
3.) In the PiKVM terminal itself, the special characters are showing just fine, it's just on the PiKVM Remote Web UI + VNC where they are not working at all. So definintly PiKVM seems to get the characters but not sending them along USB properly

If I can help in any way just let me know but atm it's really a huge problem on Swiss German layout and I think point number 2 I mentioned is very interesting. It's like PiKVM somehow turns "ALT+CTRL" into "Shift" or so.

generalmanager, 2022-01-14:
@mdevaev Have your German keyboards arrived? If not, I'd be willing to send you one. :)

I've got a related issue, which @marcelser has also described. I'm using a German (normal, non-swiss) Keyboard layout. 

Unrelated, but my browser locale is set to en-US to avoid beeing served localized content, that doesn't seem to have any impact on the bug though. It just means I have to manually switch to the German layout every time I restart my browser. Would be nice if pikvm would save the chosen keyboard layout.

If I enter the [ ] { } | \ # @ sequence described by @marcelser into the paste field of the KVM mode (DE layout is selected), it gets pasted as `ü + Ü * ' # # "`

Would be great to get this fixed, as it destroys basically all scripts when using the otherwise really useful paste tool.

mdevaev, 2022-01-14:
They've arrived, but I'm away myself right now (not in a city)

generalmanager, 2022-01-15:
I see, no worries. :) I guess the issue with pasting doesn't need a physical keyboard to debug. I have no intention of beeing pushy or annoying though, I'm really glad this project exists and is so well maintained. :) Only gripe I have is not ordering more V3s during the kickstarter. 😂

marcluer, 2022-01-24:
Subscribing to this topic as I also have keyboard layout issues with a non-US layout.

My Setup:
```
Keyboard    ->     Client OS  Win11     ->     Browser        ->     PiKVM      ->       Target
DE                 Win11                       Firefox               Pi Zero2            Pi 4
                   Lang: EN                    Lang: EN              USB        ->       USB
                   Keyboard: DE                                      CSI        ->       HDMI        
                                                                                         Lang: EN
                                                                                         Keyboard: DE   
```

When I type "-" on my keyboard in PiKVM, I receive "/" on the target.
When I plug a German keyboard directly into the target and push "-" I receive "-".

mrambossek, 2022-01-24:
i have basically the same setup as marcluer, except im using chrome. but all OS are set to EN with DE keyboard
the "normal" keys like "-" work fine for me. just the alt+gr combinations do not.

kkr0kk, 2022-02-20:
Hi, I think i know what wrong with ALTGR key with europeen keyboard, i have FR kayboard.
ALTGR key (that was the right ALT kay) is equivalent with ALT+CTRL keys.
So when we hit the ALTGR on french kayboard sa was send right ALT + CTRL code key, but on our keyboard that is not the good code combo code.
On linux when we configure keyboard (every system) we can choose how the ALTGR key interact (was default config on all linux system actually)
On FR keyboard the ALTGR key must send only the right ALT code and not send right ALT and CTRL
I think was the same problem with DE kayboard.
I hope this analyse will heelp you.

Actually my workround is to type and maintain my ALTGR key and hit once the left CTRL key and hit my "@" or "|" (when i hit the CTRL key that was disable the CTRL send code and only right ALT is enable, i see that on the virtual kayboard).

I can made a video if you want.

semool, 2022-02-24:
> Actually my workround is to type and maintain my ALTGR key and hit once the left CTRL key and hit my "@" or "|" (when i hit the CTRL key that was disable the CTRL send code and only right ALT is enable, i see that on the virtual keyboard).

Thats working also fine on a german Keyboard. Thx for the Tip.

marcelser, 2022-02-28:
> Actually my workround is to type and maintain my ALTGR key and hit once the left CTRL key and hit my "@" or "|" (when i hit the CTRL key that was disable the CTRL send code and only right ALT is enable, i see that on the virtual keyboard).

Also working fine on Swiss German keyboard as well. Press "ALTGR" and keep pressed while hitting left "CTRL" followed by the desired key. Then release "ALTGR". For multiple characters keep holding down "ALTGR"-key and then just keep hitting left "CTRL"+key for each character once you're done release "ALTGR".

marcelser, 2022-06-09:
@mdevaev any update on this annoying bug? It's been open for 1 year now. I see you're releasing lots of new stuff but no update whatsoever on this issue from what I see

mdevaev, 2022-06-09:
@marcelser Unfortunately, fixing this bug does not depend on my desire. I tried to figure out what the problem was, but I couldn't. Recently I was able to get two DE keyboards (it was difficult) to completely replicate all the external factors, and I'm going to do it soon.

mdevaev, 2022-08-22:
Please let me know if someone is ready to test the fix.

semool, 2022-08-22:
> Please let me know if someone is ready to test the fix.

I can test whatever you want. Tell me what i todo. I have a Pikvm running with the latest available version: KVMD Version: 3.137 - v2-hdmi

marcelser, 2022-08-22:
> Please let me know if someone is ready to test the fix.

I would also like to try it on Swiss German. Just let me know how to install the fix

mdevaev, 2022-08-22:
@marcelser @semool Thanks. Let's check it.
1. Edit `/etc/pacman.conf`. Find the lines:
    ```ini
    [pikvm]
    Server = https://files.pikvm.org/repos/arch//rpi4-arm
    SigLevel = Required DatabaseOptional
    ```
2. Replace the `Server` line to `Server = https://files.pikvm.org/repos/arch-testing/rpi4-arm`.
3. Perform full update: `pacman -Syu`
4. Test the web ui. Don't forget to clean the browser cache.

semool, 2022-08-22:
@mdevaev 
Tested it and its working fine now. Good Work, thank you.

mdevaev, 2022-08-22:
Glad to hear it :)

mdevaev, 2022-08-22:
Released in the stable repo.

closed 2022-08-22 by mdevaev

marcelser, 2022-08-22:
Ok, was too slow to test on the testing arch, but I just tested on stable repo and yes it works fine also on swiss german keyboard. Thank you very much for fixing this long standing bug, what a relief if you want to type something.

mdevaev, 2022-08-22:
It's a shame that it took so long, but I'm glad I was able to fix it :ok_hand: