pikvm/pikvm #382: [SOLUTION] Hardware fix on CSI to mitigate HDMI backpowering for AIMOS KVM's

arch1mede, 2021-07-07:
This requires ADVANCED soldering to perform on CSI module

[image]
[image]

Required: Diode 1N4148 (cathode of the diode faces away from pin 15)
Notes: You need to interrupt the v3.3 input and put a diode on it.  So have to break the circuit, then bridge the break with the diode. if the traces are buried that can be hard to do

OR

[image]

Replace the 0-Ohm with the above diode

closed 2021-07-08 by mdevaev

woffko, 2021-08-03:
I have different KVM but the same problem — when KVM is plugged into CSI HDMI board, Pi4 refuses to boot.  Unfortunately this solution didn't work, but I have my own. Replace the other fuse, near the HDMI socket with the same 1N4148, black side of the diode towards the HDMI connector. 
Before
[image]
After
[image]

mdevaev, 2021-08-04:
@woffko which KVM do you use? Model?

woffko, 2021-08-12:
Currently I've tried Inter-Tech KVM AS-21HA IPC HDMI, Aimos aka https://www.aliexpress.com/item/4001230679168.html?spm=a2g0s.9042311.0.0.7f094c4djuveFW and Ugreen https://www.aliexpress.com/item/32854516883.html?spm=a2g0s.9042311.0.0.27424c4dAbvWcX
I have two PCs with asrock rack boards, one is J1900D2Y and the second is E3C226D2I
J1900D2Y is very problematic, because it should be connected to the monitor or directly to the pikvm upon cold boot or you just don't have any output on HDMI, don't know what the problem is, but the second PC with E3C226D2I board works without any problem via VGA-HDMI converter.
In conclusion  Inter-Tech KVM AS-21HA IPC HDMI works, but you need to switch it manually, hotkeys are not working.
Ugreen works the same way as Inter-tech, only manual switch between sources.
Aimos that I've ordered by clicking on recommended noname Ali HDMI switch works, but I have no luck —  the PIKVM keyboard and virtual media are not working at all, tried all ports, but no luck.

ronaldlw, 2021-08-13:
> 
> 
> Currently I've tried [Inter-Tech KVM AS-21HA IPC HDMI](https://www.inter-tech.de/en/products/ipc/kvm/kvm-as-21ha), [Aimos](https://www.aliexpress.com/item/4001230679168.html?spm=a2g0s.9042311.0.0.7f094c4djuveFW) and [Ugreen](https://www.aliexpress.com/item/32854516883.html?spm=a2g0s.9042311.0.0.27424c4dAbvWcX)
> I have two PCs with asrock rack boards, one is J1900D2Y and the second is E3C226D2I
> J1900D2Y is very problematic, because it should be connected to the monitor or directly to the pikvm upon cold boot or you just don't have any output on HDMI, don't know what the problem is, but the second PC with E3C226D2I board works without any problem via VGA-HDMI converter connected to HDMI switch.
> In conclusion Inter-Tech KVM AS-21HA IPC HDMI works, but you need to switch it manually, hotkeys are not working.
> Ugreen works the same way as Inter-tech, only manual switch between sources.
> Aimos that I've ordered by clicking on recommended noname Ali HDMI switch works, but I have no luck — the PIKVM keyboard and virtual media are not working at all, tried all ports, but no luck.

I'm confused.  This sounds like what you're saying is that this "fix" doesn't work at all with any of the kvms you've tried?  I wouldn't consider "works, but requires manual switch between sources" to in any way be working, am I misunderstanding?

woffko, 2021-08-15:
> I'm confused. This sounds like what you're saying is that this "fix" doesn't work at all with any of the kvms you've tried? I wouldn't consider "works, but requires manual switch between sources" to [...]

**Backpowering  issue is fixed with all listed KVMs**, but I decided to share my experience regarding all the functions of the KVMs, not just backpowering issue. Sorry for the misunderstanding.

gotshadowbantempaccount, 2021-09-20:
Actually got same problem with Aimos and HDMI
1. when power is on on "USB 5V cable to mouse port" and then plug power to Raspberry PI original power supply 5.1V its refuse to boot. But wteh power on raspberyPI then USB power tpo HDMI its working
2. when use some cheap power supply raspbery pi only boot when is HDMi CSI 2 is disconnected
3. For me solution for now is to "power on first by remote reboot socket" 1) raspbery PI power then 2)+5V USB to mouse port
4. **Another solution will be maybe try to block power on HDMi cable same like on USB so AMIOS will be power on only from USB cable connected to keyboard port**

[image]

[image]

arch1mede, 2021-09-22:
> 
> 
> Actually got same problem with Aimos and HDMI
> 
>     1. when power is on on "USB 5V cable to mouse port" and then plug power to Raspberry PI original power supply 5.1V its refuse to boot. But wteh power on raspberyPI then USB power tpo HDMI its working
> 
>     2. when use some cheap power supply raspbery pi only boot when is HDMi CSI 2 is disconnected
> 
>     3. For me solution for now is to "power on first by remote reboot socket" 1) raspbery PI power then 2)+5V USB to mouse port
> 
>     4. **Another solution will be maybe try to block power on HDMi cable same like on USB so AMIOS will be power on only from USB cable connected to keyboard port because maybe is try to get power from raspberry pi and refuse to boot (unconvifrmed)**
> 
> 
> [image]
> 
> [image]

Your actually the first to report that. We have a bunch of active users who have AIMOS KVM's and don't have a mouse port issue like you are porting.

ronaldlw, 2021-09-22:
1) you shouldn't be applying power via the mouse port.  is that what you are saying you're doing here??
> Actually got same problem with Aimos and HDMI
>
>     1. when power is on on "USB 5V cable to mouse port" and then plug power to Raspberry PI original power supply 5.1V its refuse to boot. But wteh [...]

arch1mede, 2021-09-22:
If there is power being delivered on the mouse port, that's back power and this needs to be taken care of. Short term you will only experience odd behaviors, long term you are looking at equipment damage to the usb port or rpi or target or all 3.

gotshadowbantempaccount, 2021-09-25:
@arch1mede   @ronaldlw  problem is only with "power on" **raspebery pi can't power on** (boot)  when AIMOS in **on** before raspberry. try to disable HDMI power Pin but that not help. 

So for me now only solution is to:
1) power on first raspbery pi by "remote socket"
2) power on second socket that give power to "AIMOS" by USB

arch1mede, 2021-09-25:
> @arch1mede @ronaldlw problem is only with "power on" **raspebery pi can't power on** (boot) when AIMOS in **on** before raspberry. try to disable HDMI power Pin but that not help.
>
> So for me now [...]

This sounds suspiciously like you have a wiring issue. No one else has reported this and this def sounds like back power. Please join discord and provide pics of your setup.