pikvm/pikvm #391: During BIOS boot, keyboard free inactive/busy, no inputs detected until OS loaded

Kamaris521, 2021-07-13:
This issue is exactly like https://github.com/pikvm/pikvm/issues/146, so I am posting this as a generic inquiry versus a bug report, as I believe it is ultimately a limitation of my hardware. However, I'm hoping there is still some workaround possible that I have not been able to garner from previous posts. The previous self resolution of "You gave me the idea to plug the USB cable on one of the built in USB ports" was not a success for me.

# Describe the issue:
Keyboard inputs are not detected until system is past BIOS and OS bootloader. Keyboard icon changes to orange during initial system boot with mouseover message "keyboard free inactive/busy" until OS is fully loaded, then inputs are detected.

# Expected behavior:
During boot process, keyboard inputs (Function keys, DEL, etc.) to enter BIOS config should be detected. 

# Desktop (please complete the following information):
OS: Windows 10 20H2
Browser: Brave Version 1.26.74 Chromium: 91.0.4472.124 (Official Build) (64-bit)
Motherboard: Asus P6X58D Premium
BIOS: American Megatrends v1501 (from around 2011, no updates available since, USB Legacy is Enabled)
USB ports: 2x 3.0, 4x 2.0 (tried all, all work once OS is loaded, but not during boot)

# Pi-KVM info:
Raspberry Pi board version: Raspberry Pi 4 Model B Rev 1.4
Pi-KVM platform: v2-hdmi
Video capture type: USB HDMI dongle
KVMD: 3.8
Streamer: 4.4 (ustreamer)
Linux kernel:
Machine: armv7l
Release: 5.10.23-2-ARCH
Version: #1 SMP Thu Mar 18 18:33:35 MSK 2021

Besides the Arduino HID, Is there any other option I can try to have the ability to enter BIOS during boot from inside Pi-KVM?

mdevaev, 2021-07-13:
Hi. No, Arduino HID is the best way to achieve maximum compatibility right now. A few days ago I improved this.

semool, 2021-07-13:
I also have an old Machine (Asus P5Q-E) with crappy Bios.
Go and build an Arduino and will working fine. You can then put both USB cables in the PC (Arduino for M/K) and the other for MSD.

Kamaris521, 2021-07-13:
> I also have an old Machine (Asus P5Q-E) with crappy Bios.
> Go and build an Arduino and it will working fine. You can then put both USB cables in the PC (Arduino for M/K) and the other for MSD.

I think that will be outside the scope of what I can accomplish right now, but I'll consider it down the road if I use the application more regularly.

> Hi. No, Arduino HID is the best way to achieve maximum compatibility right now. A few days ago I improved this.

Ok ty for the response! Wonderful tool and good luck with future enhancements.

closed 2021-07-13 by Kamaris521