pikvm/pikvm #404: Pi-KVM Networking deadlock on Raspberry Pi 4 Model B Rev 1.1

atsies, 2021-07-19:
> [   18.077797] ------------[ cut here ]------------
[   18.082537] WARNING: CPU: 1 PID: 348 at net/sched/sch_generic.c:442 dev_watchdog+0x2f0/0x2f4
[   18.091097] NETDEV WATCHDOG: eth0 (bcmgenet): transmit queue 2 timed out
[   18.097881] Modules linked in: usb_f_mass_storage usb_f_hid btsdio bluetooth ecdh_generic ecc brcmfmac brcmutil bcm2835_isp(C) cfg80211 bcm2835_unicam bcm2835_codec(C) raspberrypi_hwmon bcm2835_v4l2(C) i6
[   18.153217] CPU: 1 PID: 348 Comm: rngd Tainted: G  C 5.10.23-2-ARCH pikvm/os#1
[   18.161046] Hardware name: BCM2711
[   18.164528] [<c020f10c>] (unwind_backtrace) from [<c020aed0>] (show_stack+0x10/0x14)
[   18.172362] [<c020aed0>] (show_stack) from [<c0de95c4>] (dump_stack+0xa4/0xc4)
[   18.179665] [<c0de95c4>] (dump_stack) from [<c022ae9c>] (__warn+0xec/0x148)
[   18.186699] [<c022ae9c>] (__warn) from [<c0de1868>] (warn_slowpath_fmt+0x7c/0xac)
[   18.194257] [<c0de1868>] (warn_slowpath_fmt) from [<c0cc4bcc>] (dev_watchdog+0x2f0/0x2f4)
[   18.202516] [<c0cc4bcc>] (dev_watchdog) from [<c02ae0c4>] (call_timer_fn+0x2c/0x1ac)
[   18.211759] [<c02ae0c4>] (call_timer_fn) from [<c02ae48c>] (__run_timers.part.0+0x248/0x320)
[   18.223097] [<c02ae48c>] (__run_timers.part.0) from [<c02ae59c>] (run_timer_softirq+0x38/0x68)
[   18.234677] [<c02ae59c>] (run_timer_softirq) from [<c0201208>] (__do_softirq+0x110/0x394)
[   18.245819] [<c0201208>] (__do_softirq) from [<c0230e64>] (irq_exit+0xac/0x104)
[   18.256188] [<c0230e64>] (irq_exit) from [<c028dd70>] (__handle_domain_irq+0x80/0xd0)
[   18.266998] [<c028dd70>] (__handle_domain_irq) from [<c0978b2c>] (gic_handle_irq+0x80/0x94)
[   18.278324] [<c0978b2c>] (gic_handle_irq) from [<c0200e04>] (__irq_usr+0x44/0x60)
[   18.288766] Exception stack(0xc39c7fb0 to 0xc39c7ff8)
[   18.295294] 7fa0:                                     5bb58b6b 00000038 5bb58b6b 8b8db41e
[   18.306416] 7fc0: 0000006b 00000000 5bb58b00 8b8db41e 5bb58b6b 8b8db41e 00481078 b59fe964
[   18.317575] 7fe0: ffffffe8 b59fe938 b6ebccc0 b6ebbb9c 40030010 ffffffff
[   18.325813] ---[ end trace 4958de0b6e8f1b82 ]---

Happens randomly. Ethernet is stuck after that.

atsies, 2021-07-21:
Happens from seconds to minutes after the boot.

mdevaev, 2021-07-21:
Are you sure that everything is fine with your device? Is there a shortage of food? This bug is no longer observed by anyone.

arch1mede, 2021-07-22:
@atsies I just flashed my rpi4 3x within an hour and at no time have I ever seen this. Please reflash using the latest image for your device using rpi imager. I also own 4x of these and none exhibit this behavior.

If anything, this is a hardware issue.

mdevaev, 2021-07-24:
@atsies sup?

mdevaev, 2021-07-25:
@atsies ping

atsies, 2021-07-27:
Raspberry OS has the same problem:

> Jul 27 16:00:44 raspberrypi kernel: [   29.993538] ------------[ cut here ]------------
Jul 27 16:00:44 raspberrypi kernel: [   29.993596] WARNING: CPU: 3 PID: 0 at net/sched/sch_generic.c:443 dev_watchdog+0x308/0x30c
Jul 27 16:00:44 raspberrypi kernel: [   29.993620] NETDEV WATCHDOG: eth0 (bcmgenet): transmit queue 4 timed out
Jul 27 16:00:44 raspberrypi kernel: [   29.993643] Modules linked in: cmac bnep hci_uart btbcm bluetooth ecdh_generic ecc 8021q garp stp llc brcmfmac brcmutil vc4 cec sha256_generic drm_kms_helper v3d gpu_sched drm drm_panel_orientation_quirks cfg80211 rfkill bcm2835_codec(C) raspberrypi_hwmon v4l2_mem2mem bcm2835_v4l2(C) bcm2835_isp(C) snd_soc_core bcm2835_mmal_vchiq(C) videobuf2_vmalloc snd_compress videobuf2_dma_contig snd_pcm_dmaengine snd_bcm2835(C) videobuf2_memops videobuf2_v4l2 snd_pcm snd_timer videobuf2_common videodev snd vc_sm_cma(C) mc rpivid_mem syscopyarea sysfillrect sysimgblt fb_sys_fops nvmem_rmem backlight uio_pdrv_genirq uio ip_tables x_tables ipv6
Jul 27 16:00:44 raspberrypi kernel: [   29.994897] CPU: 3 PID: 0 Comm: swapper/3 Tainted: G         C        5.10.17-v7l+ #1414
Jul 27 16:00:44 raspberrypi kernel: [   29.994914] Hardware name: BCM2711
Jul 27 16:00:44 raspberrypi kernel: [   29.994927] Backtrace: 
Jul 27 16:00:44 raspberrypi kernel: [   29.994962] [<c0b63050>] (dump_backtrace) from [<c0b633e4>] (show_stack+0x20/0x24)
Jul 27 16:00:44 raspberrypi kernel: [   29.994978]  r7:ffffffff r6:00000000 r5:60000113 r4:c12e69fc
Jul 27 16:00:44 raspberrypi kernel: [   29.994997] [<c0b633c4>] (show_stack) from [<c0b67774>] (dump_stack+0xcc/0xf8)
Jul 27 16:00:44 raspberrypi kernel: [   29.995022] [<c0b676a8>] (dump_stack) from [<c0220bac>] (__warn+0xfc/0x114)
Jul 27 16:00:44 raspberrypi kernel: [   29.995039]  r10:c133c8b8 r9:00000009 r8:c0a46634 r7:000001bb r6:00000009 r5:c0a46634
Jul 27 16:00:44 raspberrypi kernel: [   29.995053]  r4:c0ea7698 r3:c1205094
Jul 27 16:00:44 raspberrypi kernel: [   29.995072] [<c0220ab0>] (__warn) from [<c0b63b88>] (warn_slowpath_fmt+0xa4/0xd8)
Jul 27 16:00:44 raspberrypi kernel: [   29.995086]  r7:000001bb r6:c0ea7698 r5:c1205048 r4:c0ea765c
Jul 27 16:00:44 raspberrypi kernel: [   29.995107] [<c0b63ae8>] (warn_slowpath_fmt) from [<c0a46634>] (dev_watchdog+0x308/0x30c)
Jul 27 16:00:44 raspberrypi kernel: [   29.995122]  r9:eff47540 r8:c2280000 r7:c1203d00 r6:c21dac00 r5:c22802a8 r4:00000004
Jul 27 16:00:44 raspberrypi kernel: [   29.995141] [<c0a4632c>] (dev_watchdog) from [<c02aa600>] (call_timer_fn+0x40/0x1bc)
Jul 27 16:00:44 raspberrypi kernel: [   29.995154]  r8:c193be04 r7:ffff9688 r6:c0a4632c r5:00000100 r4:c22802a8
Jul 27 16:00:44 raspberrypi kernel: [   29.995172] [<c02aa5c0>] (call_timer_fn) from [<c02abc20>] (run_timer_softirq+0x5b0/0x698)
Jul 27 16:00:44 raspberrypi kernel: [   29.995184]  r8:c193be04 r7:00000000 r6:c22802a8 r5:ffff9688 r4:00000000
Jul 27 16:00:44 raspberrypi kernel: [   29.995202] [<c02ab670>] (run_timer_softirq) from [<c0201508>] (__do_softirq+0x198/0x49c)
Jul 27 16:00:44 raspberrypi kernel: [   29.995220]  r10:00000082 r9:ffffe000 r8:c1810800 r7:00000100 r6:00000001 r5:00000002
Jul 27 16:00:44 raspberrypi kernel: [   29.995233]  r4:c1203084
Jul 27 16:00:44 raspberrypi kernel: [   29.995253] [<c0201370>] (__do_softirq) from [<c0227474>] (irq_exit+0xd0/0xf8)
Jul 27 16:00:44 raspberrypi kernel: [   29.995269]  r10:c0e1edfc r9:c193a000 r8:c1810800 r7:00000001 r6:00000000 r5:00000000
Jul 27 16:00:44 raspberrypi kernel: [   29.995283]  r4:ffffe000
Jul 27 16:00:44 raspberrypi kernel: [   29.995307] [<c02273a4>] (irq_exit) from [<c0287540>] (__handle_domain_irq+0x70/0xc4)
Jul 27 16:00:44 raspberrypi kernel: [   29.995319]  r5:00000000 r4:c1094d10
Jul 27 16:00:44 raspberrypi kernel: [   29.995338] [<c02874d0>] (__handle_domain_irq) from [<c020135c>] (gic_handle_irq+0x90/0xa4)
Jul 27 16:00:44 raspberrypi kernel: [   29.995354]  r9:c193a000 r8:c1094d1c r7:c193bf30 r6:f081400c r5:f0814000 r4:c1205b3c
Jul 27 16:00:44 raspberrypi kernel: [   29.995370] [<c02012cc>] (gic_handle_irq) from [<c0200abc>] (__irq_svc+0x5c/0x7c)
Jul 27 16:00:44 raspberrypi kernel: [   29.995382] Exception stack(0xc193bf30 to 0xc193bf78)
Jul 27 16:00:44 raspberrypi kernel: [   29.995399] bf20:                                     00000000 00016e14 eff4f2c4 c021ac00
Jul 27 16:00:44 raspberrypi kernel: [   29.995414] bf40: ffffe000 c120509c c12050e4 00000008 00000001 c133ba3f c0e1edfc c193bf8c
Jul 27 16:00:44 raspberrypi kernel: [   29.995427] bf60: c193bf90 c193bf80 c02088c0 c02088c4 60000013 ffffffff
Jul 27 16:00:44 raspberrypi kernel: [   29.995441]  r9:c193a000 r8:00000001 r7:c193bf64 r6:ffffffff r5:60000013 r4:c02088c4
Jul 27 16:00:44 raspberrypi kernel: [   29.995460] [<c020887c>] (arch_cpu_idle) from [<c0b72ef0>] (default_idle_call+0x4c/0x118)
Jul 27 16:00:44 raspberrypi kernel: [   29.995476] [<c0b72ea4>] (default_idle_call) from [<c02587d4>] (do_idle+0x118/0x168)
Jul 27 16:00:44 raspberrypi kernel: [   29.995492] [<c02586bc>] (do_idle) from [<c0258af8>] (cpu_startup_entry+0x28/0x30)
Jul 27 16:00:44 raspberrypi kernel: [   29.995506]  r10:00000000 r9:410fd083 r8:00007000 r7:c13543a0 r6:30c0387d r5:00000003
Jul 27 16:00:44 raspberrypi kernel: [   29.995516]  r4:00000091 r3:60000093
Jul 27 16:00:44 raspberrypi kernel: [   29.995534] [<c0258ad0>] (cpu_startup_entry) from [<c020f7b8>] (secondary_start_kernel+0x16c/0x178)
Jul 27 16:00:44 raspberrypi kernel: [   29.995550] [<c020f64c>] (secondary_start_kernel) from [<002018ac>] (0x2018ac)
Jul 27 16:00:44 raspberrypi kernel: [   29.995561]  r5:00000000 r4:018ceac0
Jul 27 16:00:44 raspberrypi kernel: [   29.995583] ---[ end trace b07174eb63dd409b ]---

mdevaev, 2021-07-27:
Okay, I see. I recommend you to post this issue to official [RPi bugtracker](https://github.com/raspberrypi/linux/issues) because this is not a Pi-KVM or Arch problem. Moreover, since the problem is specific to you, it is possible that you have a defective board, this sometimes happens (I have encountered this). RPi developers can refute (and fix) this can be confirmed, and then most likely your device will be replaced under warranty.

mdevaev, 2021-07-27:
Unfortunately, I can't help you with anything else here.

closed 2021-07-27 by mdevaev

atsies, 2021-07-28:
Another new Rpi 4 Rev 1.2 ordered. The same issue.
Reported:
https://github.com/raspberrypi/linux/issues/4485

mdevaev, 2021-07-28:
Thanks. I've subscribed to that issue.

atsies, 2021-08-30:
Ordered a 3'rd RPi4 B rev. 1.2 . The same issue.
It looks like the network traffic is the cause of the problem. After switching to another network the RPi ethernet stopped hanging.
On switching back to the 1st network, it hangs again regularly. Unfortunately, I am not yet able to record the traffic on Ethernet port to see any anomalies.