pikvm/pikvm #407: Bug: ezcoo EZ-SW41HA-KVMU3L

alteoxCOM, 2021-07-20:
Hi, I connected the HDMI switch like described and made the configuration changes.
The GPIO menu shows up and indicated me which port is active, but the "click" buttons do not switch the input.

[image]

How can I debug to solve the problem? Or does someone has an idea?

Thanks.

mdevaev, 2021-07-20:
Show the config please.

mdevaev, 2021-07-20:
Also try this: https://github.com/pikvm/pikvm/blob/master/pages/ezcoo.md#additional-step-for-the-usb30-version

alteoxCOM, 2021-07-21:
Here the config:

`[root@pikvm ~]# more /etc/kvmd/override.yaml
####################################################################
#                                                                  #
# Override Pi-KVM system settings. This file uses the YAML syntax. #
#                                                                  #
#    https://github.com/pikvm/pikvm/blob/master/pages/config.md    #
#                                                                  #
# All overridden parameters will be applied AFTER other configs    #
# and "!include" directives and BEFORE validation.                 #
# Not: Sections should be combined under shared keys.              #
#                                                                  #
####################################################################
#
#########
# Wrong #
#########
#kvmd:
#    gpio:
#        drivers: ...
#kvmd:
#    gpio:
#        scheme: ...
#
###########
# Correct #
###########
#kvmd:
#    gpio:
#        drivers: ...
#        scheme: ...
#
###########
# Example #
###########
vnc:
#    # See https://github.com/pikvm/pikvm/blob/master/pages/vnc.md
#    keymap: /usr/share/kvmd/keymaps/ru # Set russian keymap
    auth:
        vncauth:
            enabled: true # Enable auth via /etc/kvmd/vncpasswd
kvmd:
    gpio:
        drivers:
            ez:
                type: ezcoo
                protocol: 2
                device: /dev/ttyUSB0
        scheme:
            ch0_led:
                driver: ez
                pin: 0
                mode: input
            ch1_led:
                driver: ez
                pin: 1
                mode: input
            ch2_led:
                driver: ez
                pin: 2
                mode: input
            ch3_led:
                driver: ez
                pin: 3
                mode: input
            ch0_button:
                driver: ez
                pin: 0
                mode: output
                switch: false
            ch1_button:
                driver: ez
                pin: 1
                mode: output
                switch: false
            ch2_button:
                driver: ez
                pin: 2
                mode: output
                switch: false
            ch3_button:
                driver: ez
                pin: 3
                mode: output
                switch: false
        view:
            table:
                - ["#Input 1", ch0_led, ch0_button]
                - ["#Input 2", ch1_led, ch1_button]
                - ["#Input 3", ch2_led, ch2_button]
                - ["#Input 4", ch3_led, ch3_button]`

I also choose the correct USB device:

`[root@pikvm ~]# ls /dev/ttyUSB*
/dev/ttyUSB0`

I  tried to connect the management USB cable to one of the USB2 and USB3 ports. Same result.

`[128252.691615] usb 1-1.1: new full-speed USB device number 4 using xhci_hcd
[128252.827634] usb 1-1.1: New USB device found, idVendor=1a86, idProduct=7523, bcdDevice= 2.64
[128252.837844] usb 1-1.1: New USB device strings: Mfr=0, Product=2, SerialNumber=0
[128252.847075] usb 1-1.1: Product: USB Serial
[128252.858208] ch341 1-1.1:1.0: ch341-uart converter detected
[128252.869686] usb 1-1.1: ch341-uart converter now attached to ttyUSB0`

mdevaev, 2021-07-23:
If you manually switch channels using the buttons on the device body, do you see changes in the Pi-KVM menu?

alteoxCOM, 2021-07-23:
Yes this works.
Thats the strange point, the USB connection is there.

mdevaev, 2021-07-23:
Just to clarify: have you tried both protocols? I mean without `protocol: 2` too.

mdevaev, 2021-07-24:
If both protocols do not work, I will ask you to grant me remote access to your Pi-KVM for debugging. This problem will be solved.

alteoxCOM, 2021-07-24:
Yes I tried the both protocols.

I sent you an e-mail with the login details.

Thank you in advance

mdevaev, 2021-07-25:
An update was required to support protocol=2

closed 2021-07-25 by mdevaev

QNimbus, 2022-02-06:
> An update was required to support protocol=2

I have the exact same problem as @alteoxCOM - can you please explain what needs to be updated to support `protocol: 2` in the override.yaml file?

arch1mede, 2022-02-07:
> > An update was required to support protocol=2
> 
> I have the exact same problem as @alteoxCOM - can you please explain what needs to be updated to support `protocol: 2` in the override.yaml file?

This issue was resolved, bad syntax in the yaml file.