pikvm/pikvm #410: Cannot use usb commands HP BIOS (EliteBook, ZBook..)

anjovanovic, 2021-07-22:
Hi,
HP BIOS ignores usb (mouse, keyboard and MASS STORAGE DEVICE) but after booting in OS everything going fine.
For example with TinyPilot (Raspios) everything is fine in BIOS, I am able to use mouse and keyboard.

I don't exactly what is problem, but firstly can we  build pikvm image with raspios, maybe there is problem with drivers in Arch Linux?
There is error too:
configfs-gadget gadget: error in submission: ep3out --> -11

mdevaev, 2021-07-22:
Arch and RaspOS use one and the same kernel. Are you using the same hardware on the Raspberry Pi side?

mdevaev, 2021-07-22:
Maybe it's mass storage. If TP does not have it, then you can disable it from Pi-KVM and then everything will work.

To do this run `rw` command, edit `/etc/kvmd/override.yaml` and append:
```
kvmd:
    msd:
        type: disabled
```
And run `reboot`.

anjovanovic, 2021-07-22:
> Arch and RaspOS use one and the same kernel. Are you using the same hardware on the Raspberry Pi side?

Yes, same HW.

anjovanovic, 2021-07-22:
> Maybe it's mass storage. If TP does not have it, then you can disable it from Pi-KVM and then everything will work.
> 
> To do this run `rw` command, edit `/etc/kvmd/override.yaml` and append:
> 
> ```
> kvmd:
>     msd:
>         type: disabled
> ```
> 
> And run `reboot`.

I will check this tomorrow. Thank you

anjovanovic, 2021-07-23:
> I will check this tomorrow. Thank you

Yes, this sloves problem. Thanks!

closed 2021-07-23 by anjovanovic

mdevaev, 2021-07-23:
I'm glad to hear it! Please tell me the laptop models. Maybe I'll buy or rent this to make Mass Storage work.

anjovanovic, 2021-07-23:
HP EliteBook 840 G5 or G6, HP ZBook 15 G3. we also cannot use video stream in BIOS mode with ZBook, ELiteBook works.
We use CSI adapter https://www.aliexpress.com/item/4000102166176.html