pikvm/pikvm #415: password for kvmd-webterm

connected201, 2021-07-26:
Hi, i want to install [duplicati](https://xo.tc/installing-duplicati-on-an-arch-linux-laptop.html) from this tutorial, but when i want to run `makepkg -si`, kvmd-webterm ask me a password, but "root" and "admin" not give me root access from web console:
[image]

from ssh, i logging as root user by "root"username and password. When i want to run `makepkg -si` i receive this:

[image]
i changed default password, but not helped.

connected201, 2021-07-26:
i able ti install with these steps:
as root from ssh i installed dependencies for duplicati and git
pacman -Sy gtk-sharp-2 mono git
git clone https://aur.archlinux.org/duplicati-latest.git

as user from web interface:
cd duplicati-latest
makepkg # wait for installation complete

but i cannot run it because as root i receive
[image]
as user, i need to enter password, but admin and root password not valid
[image]

mdevaev, 2021-07-27:
kvmd-webterm does not have a password. To switch to root, use `su -`.
To build package switch to root then change directory (`cd`) and build packages using `sudo -u kvmd-webterm makepkg ...`.

connected201, 2021-07-27:
thanks, i able to install with sudo -u kvmd-webterm makepkg
but i cannot run 
```
systemctl enable duplicati.service
systemctl start duplicati.service
```
also i tried with
`sudo -u kvmd-webterm` but no results, any idea?

[image]

connected201, 2021-07-27:
also, i nstalled AUR package manager, but when i want to run `sudo -u kvmd-webterm yay`
i received
`[kvmd-webterm@pikvm(g:master) g:/]$ sudo -u kvmd-webterm yay
kvmd-webterm is not in the sudoers file.  This incident will be reported.`
i just want to install some additional tools, but i unable to run it!

mdevaev, 2021-07-27:
under kvmd-webterm user you don't need to make sudo again

connected201, 2021-07-28:
i had problem when i want to install something,
```
[root@pikvm(g:master) g:/]# sudo -u kvmd-webterm makepkg -si
==> ERROR: Failed to create the directory $BUILDDIR (/root/yay).
```
 but i resolve with folder rights
`chmod 777 /root/appfolder`

connected201, 2021-07-28:
right now i have problem, i cannot use apps installed from other sources. only apps installed for pacman can work (`pacman -S appname`)
```
[kvmd-webterm@pikvm(g:master) g:/]$ yay
bash: yay: command not found
[kvmd-webterm@pikvm(g:master) g:/]$ su -
Password: 
[root@pikvm ~]# yay
-bash: yay: command not found
[root@pikvm ~]# 
```

mdevaev, 2021-07-28:
makepkg does not install the package, use `pacman -U ...` on built package

connected201, 2021-07-28:
All working great, thanks for help!

closed 2021-07-28 by connected201

mdevaev, 2021-07-28:
Welcome!