pikvm/pikvm #417: how to extend /dev/mmcblk0p2?

connected201, 2021-07-29:
Hi, i need more space for `/dev/mmcblk0p2` . No left space available because i make some experiments and 5.6G is not enough, maybe i's possible via gparted?.
```
[root@pikvm ~]# df -h
df: /root/mounted: Transport endpoint is not connected
Filesystem      Size  Used Avail Use% Mounted on
dev             332M     0  332M   0% /dev
run             401M   17M  385M   5% /run
/dev/mmcblk0p2  5.6G  5.6G     0 100% /
tmpfs           401M   88K  401M   1% /dev/shm
tmpfs           401M     0  401M   0% /tmp
tmpfs           401M     0  401M   0% /var/lib/dhcpcd
/dev/mmcblk0p1  256M   41M  216M  16% /boot
tmpfs           401M     0  401M   0% /var/lib/dhclient
/dev/mmcblk0p3   52G   32K   52G   1% /var/lib/kvmd/msd
tmpfs           401M     0  401M   0% /var/lib/private
tmpfs           401M  240K  401M   1% /var/lib/systemd
tmpfs           401M   12K  401M   1% /var/log
tmpfs           401M     0  401M   0% /var/tmp
/dev/sda1        29G   26G  3.4G  89% /root/usb
[root@pikvm ~]#
```

mdevaev, 2021-07-29:
As usual in linux, this is difficult. I recommend to remove old packages: `rm -rf /var/cache/pacman/pkg`.

connected201, 2021-07-29:
i just connect microsd card to my pc, launch live linux, open gparted (from live cd) and resize first `/dev/mmcblk0p3   52G` and after resize `/dev/mmcblk0p2  5.6G`
and all working great.
```
[root@pikvm ~]# df -h
Filesystem      Size  Used Avail Use% Mounted on
dev             332M     0  332M   0% /dev
run             401M  8.6M  393M   3% /run
/dev/mmcblk0p2   55G   20G   32G  39% /
tmpfs           401M   88K  401M   1% /dev/shm
tmpfs           401M     0  401M   0% /tmp
/dev/mmcblk0p1  256M   41M  216M  16% /boot
tmpfs           401M     0  401M   0% /var/lib/dhclient
tmpfs           401M     0  401M   0% /var/lib/dhcpcd
tmpfs           401M     0  401M   0% /var/lib/private
tmpfs           401M  8.0K  401M   1% /var/lib/systemd
tmpfs           401M  8.0K  401M   1% /var/log
tmpfs           401M     0  401M   0% /var/tmp
/dev/mmcblk0p3  2.9G -806M  3.7G    - /var/lib/kvmd/msd
[root@pikvm ~]#
```

closed 2021-07-29 by connected201

mdevaev, 2021-07-29:
Yea, why not.