pikvm/pikvm #422: Please support "X-Forwarded-For header" for proxied connections

Taomyn, 2021-08-09:
**Is your feature request related to a problem? Please describe.**
Trying to get HAProxy to work but PiKVM always logs incoming connection as the IP of firewall not remote machine i.e. in my case they are all 192.168.1.1 no matter where I am connected. Would be useful for security purposes as well.

**Describe the solution you'd like**
Check the X-Forwarded-For header and use the contained IP

mdevaev, 2021-08-09:
This header is [already](https://github.com/pikvm/kvmd/blob/master/configs/nginx/loc-proxy.conf) used and forwarded.

Taomyn, 2021-08-09:
> This header is [already](https://github.com/pikvm/kvmd/blob/master/configs/nginx/loc-proxy.conf) used and forwarded.

Strange, as in my testing today all the connections showed the firewall IP - I will check again tomorrow as I won't be anywhere I can check from external again. I know this works because I have other internal sites that rely on the correct external IP and they are working - will check that again as well.

Taomyn, 2021-08-10:
Hi again, so I am now at the office and checking my PiKVM at home, and this is what the log view shows me. Looks like all the connections are coming in from 192.168.1.1 which is my firewall that proxies the connection to the PiKVM:

```
[2021-08-10 11:23:16 kvmd.service] --- kvmd.apps.kvmd.auth               INFO --- Authorized user 'admin' via auth service 'htpasswd'
[2021-08-10 11:23:16 kvmd.service] --- aiohttp.access                    INFO --- [- / 192.168.1.1] 'POST /auth/login HTTP/1.0' => 200; size=296 --- referer='https://pikvm.mydomain.com/login/'; user_agent='Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/92.0.4515.131 Safari/537.36'
[2021-08-10 11:23:16 kvmd.service] --- aiohttp.access                    INFO --- [admin (token) / -] 'GET /auth/check HTTP/1.0' => 200; size=199 --- referer='https://pikvm.mydomain.com/login/'; user_agent='Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/92.0.4515.131 Safari/537.36'
[2021-08-10 11:23:16 kvmd.service] --- aiohttp.access                    INFO --- [admin (token) / -] 'GET /auth/check HTTP/1.0' => 200; size=199 --- referer='https://pikvm.mydomain.com/login/'; user_agent='Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/92.0.4515.131 Safari/537.36'
[2021-08-10 11:23:16 kvmd.service] --- aiohttp.access                    INFO --- [admin (token) / 192.168.1.1] 'GET /info?fields=meta,extras HTTP/1.0' => 200; size=2105 --- referer='https://pikvm.mydomain.com/'; user_agent='Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/92.0.4515.131 Safari/537.36'
[2021-08-10 11:23:16 kvmd.service] --- aiohttp.access                    INFO --- [admin (token) / -] 'GET /auth/check HTTP/1.0' => 200; size=199 --- referer='https://pikvm.mydomain.com/'; user_agent='Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/92.0.4515.131 Safari/537.36'
[2021-08-10 11:23:19 kvmd.service] --- aiohttp.access                    INFO --- [admin (token) / -] 'GET /auth/check HTTP/1.0' => 200; size=199 --- referer='https://pikvm.mydomain.com/'; user_agent='Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/92.0.4515.131 Safari/537.36'
[2021-08-10 11:23:19 kvmd.service] --- aiohttp.access                    INFO --- [admin (token) / -] 'GET /auth/check HTTP/1.0' => 200; size=199 --- referer='https://pikvm.mydomain.com/'; user_agent='Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/92.0.4515.131 Safari/537.36'
[2021-08-10 11:23:19 kvmd.service] --- aiohttp.access                    INFO --- [admin (token) / -] 'GET /auth/check HTTP/1.0' => 200; size=199 --- referer='https://pikvm.mydomain.com/'; user_agent='Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/92.0.4515.131 Safari/537.36'
[2021-08-10 11:23:20 kvmd.service] --- aiohttp.access                    INFO --- [admin (token) / 192.168.1.1] 'GET /auth/check HTTP/1.0' => 200; size=199 --- referer='https://pikvm.mydomain.com/kvm/'; user_agent='Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/92.0.4515.131 Safari/537.36'
[2021-08-10 11:23:20 kvmd.service] --- aiohttp.access                    INFO --- [admin (token) / 192.168.1.1] 'GET /ws HTTP/1.0' => 400; size=232 --- referer='-'; user_agent='Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/92.0.4515.131 Safari/537.36'
[2021-08-10 11:23:21 kvmd.service] --- aiohttp.access                    INFO --- [admin (token) / 192.168.1.1] 'GET /auth/check HTTP/1.0' => 200; size=199 --- referer='https://pikvm.mydomain.com/kvm/'; user_agent='Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/92.0.4515.131 Safari/537.36'
[2021-08-10 11:23:21 kvmd.service] --- aiohttp.access                    INFO --- [admin (token) / 192.168.1.1] 'GET /ws HTTP/1.0' => 400; size=232 --- referer='-'; user_agent='Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/92.0.4515.131 Safari/537.36'
[2021-08-10 11:23:22 kvmd.service] --- aiohttp.access                    INFO --- [admin (token) / 192.168.1.1] 'GET /auth/check HTTP/1.0' => 200; size=199 --- referer='https://pikvm.mydomain.com/kvm/'; user_agent='Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/92.0.4515.131 Safari/537.36'
[2021-08-10 11:23:23 kvmd.service] --- aiohttp.access                    INFO --- [admin (token) / 192.168.1.1] 'GET /ws HTTP/1.0' => 400; size=232 --- referer='-'; user_agent='Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/92.0.4515.131 Safari/537.36'
[2021-08-10 11:23:24 kvmd.service] --- aiohttp.access                    INFO --- [admin (token) / 192.168.1.1] 'GET /auth/check HTTP/1.0' => 200; size=199 --- referer='https://pikvm.mydomain.com/kvm/'; user_agent='Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/92.0.4515.131 Safari/537.36'
[2021-08-10 11:23:24 kvmd.service] --- aiohttp.access                    INFO --- [admin (token) / 192.168.1.1] 'GET /ws HTTP/1.0' => 400; size=232 --- referer='-'; user_agent='Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/92.0.4515.131 Safari/537.36'
[... cut ...] 64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/92.0.4515.131 Safari/537.36'
[2021-08-10 11:23:35 kvmd.service] --- aiohttp.access                    INFO --- [admin (token) / 192.168.1.1] 'GET /auth/check HTTP/1.0' => 200; size=199 --- referer='https://pikvm.mydomain.com/kvm/'; user_agent='Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/92.0.4515.131 Safari/537.36'

```

Here is what a phpinfo test shows on another server within my network:
[image]

mdevaev, 2021-08-10:
In this case, the problem is in your firewall, isn't it?

mdevaev, 2021-08-10:
Hm, I have an idea. Change access-log format using `/etc/kvmd/override.yaml`:
```yaml
kvmd:
    server:
        access_log_format: "[%P / %{X-Forwarded-For}i] '%r' => %s; size=%b --- referer='%{Referer}i'; user_agent='%{User-Agent}i"
```

Taomyn, 2021-08-10:
I tried it by adding that to my file as I already have things in there but it didn't work - I thought maybe because there could be single quote missing near the end, but still nothing. This is my file minus the starting comments:

```
kvmd:
    atx:
        type: disabled
    gpio:
        drivers:
            tes:
                type: tesmart
                host: 192.168.1.26
                port: 5000
        scheme:
            server0_led:
                driver: tes
                pin: 0
                mode: input
            server0_switch:
                driver: tes
                pin: 0
                mode: output
                switch: false    
            server1_led:
                driver: tes
                pin: 1
                mode: input
            server1_switch:
                driver: tes
                pin: 1
                mode: output
                switch: false    
            server2_led:
                driver: tes
                pin: 2
                mode: input
            server2_switch:
                driver: tes
                pin: 2
                mode: output
                switch: false    
            server3_led:
                driver: tes
                pin: 3
                mode: input
            server3_switch:
                driver: tes
                pin: 3
                mode: output
                switch: false    
            server4_led:
                driver: tes
                pin: 4
                mode: input
            server4_switch:
                driver: tes
                pin: 4
                mode: output
                switch: false    
            server5_led:
                driver: tes
                pin: 5
                mode: input
            server5_switch:
                driver: tes
                pin: 5
                mode: output
                switch: false    
            server6_led:
                driver: tes
                pin: 6
                mode: input
            server6_switch:
                driver: tes
                pin: 6
                mode: output
                switch: false    
            server7_led:
                driver: tes
                pin: 7
                mode: input
            server7_switch:
                driver: tes
                pin: 7
                mode: output
                switch: false    
        view:
            header:
                title: KVM
            table:
                - ["TESMART KVM Switch"]
                - []
                - ["#ABE", server0_led, server0_switch|Switch]
                - ["#LISA", server1_led, server1_switch|Switch]
                - ["#RALPH", server2_led, server2_switch|Switch]
                - ["#ZIFF", server3_led, server3_switch|Switch]
                - ["#HIBBERT", server4_led, server4_switch|Switch]
                - ["#Server 6", server5_led, server5_switch|Switch]
                - ["#Server 7", server6_led, server6_switch|Switch]
                - ["#Server 8", server7_led, server7_switch|Switch]
    hid:
        mouse_alt:
            device: /dev/kvmd-hid-mouse-alt
    server:
        access_log_format: "[%P / %{X-Forwarded-For}i] '%r' => %s; size=%b --- referer='%{Referer}i'; user_agent='%{User-Agent}i'"

```

Taomyn, 2021-08-10:
Sorry, ignore that, it's working now - I forgot to reboot, d'oh!

```
[2021-08-10 12:32:23 kvmd.service] --- aiohttp.access                    INFO --- [admin (token) / 193.91.40.113, 192.168.1.1] 'GET /auth/check HTTP/1.0' => 200; size=199 --- referer='https://pikvm.mydomain.com/kvm/'; user_agent='Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/92.0.4515.131 Safari/537.36'
[2021-08-10 12:32:23 kvmd.service] --- aiohttp.access                    INFO --- [admin (token) / 193.91.40.113, 192.168.1.1] 'GET /ws HTTP/1.0' => 400; size=232 --- referer='-'; user_agent='Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/92.0.4515.131 Safari/537.36'
[2021-08-10 12:32:25 kvmd.service] --- aiohttp.access                    INFO --- [admin (token) / 193.91.40.113, 192.168.1.1] 'GET /auth/check HTTP/1.0' => 200; size=199 --- referer='https://pikvm.mydomain.com/kvm/'; user_agent='Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/92.0.4515.131 Safari/537.36'
[2021-08-10 12:32:26 kvmd.service] --- aiohttp.access                    INFO --- [admin (token) / 193.91.40.113, 192.168.1.1] 'GET /ws HTTP/1.0' => 400; size=232 --- referer='-'; user_agent='Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/92.0.4515.131 Safari/537.36'
[2021-08-10 12:32:27 kvmd.service] --- aiohttp.access                    INFO --- [admin (token) / 193.91.40.113, 192.168.1.1] 'GET /ws HTTP/1.0' => 400; size=232 --- referer='-'; user_agent='Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/92.0.4515.131 Safari/537.36'
[2021-08-10 12:32:28 kvmd.service] --- aiohttp.access                    INFO --- [admin (token) / 193.91.40.113, 192.168.1.1] 'GET /auth/check HTTP/1.0' => 200; size=199 --- referer='https://pikvm.mydomain.com/kvm/'; user_agent='Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/92.0.4515.131 Safari/537.36'
[2021-08-10 12:32:30 kvmd.service] --- aiohttp.access                    INFO --- [admin (token) / 193.91.40.113, 192.168.1.1] 'GET /auth/check HTTP/1.0' => 200; size=199 --- referer='https://pikvm.mydomain.com/kvm/'; user_agent='Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/92.0.4515.131 Safari/537.36'
[2021-08-10 12:32:30 kvmd.service] --- aiohttp.access                    INFO --- [admin (token) / 193.91.40.113, 192.168.1.1] 'GET /ws HTTP/1.0' => 400; size=232 --- referer='-'; user_agent='Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/92.0.4515.131 Safari/537.36'
[2021-08-10 12:32:32 kvmd.service] --- aiohttp.access                    INFO --- [admin (token) / 193.91.40.113, 192.168.1.1] 'GET /auth/check HTTP/1.0' => 200; size=199 --- referer='https://pikvm.mydomain.com/kvm/'; user_agent='Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/92.0.4515.131 Safari/537.36'
[2021-08-10 12:32:34 kvmd.service] --- aiohttp.access                    INFO --- [admin (token) / 193.91.40.113, 192.168.1.1] 'GET /ws HTTP/1.0' => 400; size=232 --- referer='-'; user_agent='Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/92.0.4515.131 Safari/537.36'
[2021-08-10 12:32:34 kvmd.service] --- aiohttp.access                    INFO --- [admin (token) / 193.91.40.113, 192.168.1.1] 'GET /ws HTTP/1.0' => 400; size=232 --- referer='-'; user_agent='Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/92.0.4515.131 Safari/537.36'
[2021-08-10 12:32:36 kvmd.service] --- aiohttp.access                    INFO --- [admin (token) / 193.91.40.113, 192.168.1.1] 'GET /auth/check HTTP/1.0' => 200; size=199 --- referer='https://pikvm.mydomain.com/kvm/'; user_agent='Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/92.0.4515.131 Safari/537.36'
[2021-08-10 12:32:37 kvmd.service] --- aiohttp.access                    INFO --- [admin (token) / 193.91.40.113, 192.168.1.1] 'GET /auth/check HTTP/1.0' => 200; size=199 --- referer='https://pikvm.mydomain.com/kvm/'; user_agent='Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/92.0.4515.131 Safari/537.36'
[2021-08-10 12:32:37 kvmd.service] --- aiohttp.access                    INFO --- [admin (token) / 193.91.40.113, 192.168.1.1] 'GET /ws HTTP/1.0' => 400; size=232 --- referer='-'; user_agent='Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/92.0.4515.131 Safari/537.36'
[2021-08-10 12:32:39 kvmd.service] --- aiohttp.access                    INFO --- [admin (token) / 193.91.40.113, 192.168.1.1] 'GET /auth/check HTTP/1.0' => 200; size=199 --- referer='https://pikvm.mydomain.com/kvm/'; user_agent='Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/92.0.4515.131 Safari/537.36'
[2021-08-10 12:32:39 kvmd.service] --- aiohttp.access                    INFO --- [admin (token) / 193.91.40.113, 192.168.1.1] 'GET /ws HTTP/1.0' => 400; size=232 --- referer='-'; user_agent='Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/92.0.4515.131 Safari/537.36'
[... cut ...]
[2021-08-10 12:32:47 kvmd.service] --- aiohttp.access                    INFO --- [admin (token) / 193.91.40.113, 192.168.1.1] 'GET /auth/check HTTP/1.0' => 200; size=199 --- referer='https://pikvm.mydomain.com/kvm/'; user_agent='Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/92.0.4515.131 Safari/537.36'
[2021-08-10 12:32:48 kvmd.service] --- aiohttp.access                    INFO --- [admin (token) / 193.91.40.113, 192.168.1.1] 'GET /auth/check HTTP/1.0' => 200; size=199 --- referer='https://pikvm.mydomain.com/kvm/'; user_agent='Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/92.0.4515.131 Safari/537.36'
```

mdevaev, 2021-08-10:
So, success?)

Taomyn, 2021-08-10:
> So, success?)

Yes, and I now know why my HAProxy would not work - PiKVM does not like HTTPS/2, had to enable HTTPS/1.1 on its entry in HAProxy and bang, I immediately got connected and can see my screen. Only MJPEG at the moment, need work out why WebRTC isn't. I though Guacamole used WebRTC as well it works so not sure yet why PiKVM isn't.......

mdevaev, 2021-08-12:
Perhaps you need to pass `/api/ws` and `/janus/ws` as websocket location.

Also you can play with proxy params in `/etc/kvmd/nginx/loc-proxy.conf` to override `X-Real-IP` behaviour: https://serverfault.com/questions/314574/nginx-real-ip-header-and-x-forwarded-for-seems-wrong

I'm closing this because the problem has been solved.

closed 2021-08-12 by mdevaev