pikvm/pikvm #424: Add option to download .iso file from internet

connected201, 2021-08-10:
hi, please add field where i can put direct link to .iso image like `https://releases.ubuntu.com/20.04.2.0/ubuntu-20.04.2.0-desktop-amd64.iso` and you just make wget and save .iso locally, also will need download progress, and download speed, thanks.

mdevaev, 2021-08-10:
Already implemented, please update OS to get this. In web UI press "Select image to upload".

mdevaev, 2021-08-10:
Duplicate of #148

closed 2021-08-10 by mdevaev

connected201, 2021-10-13:
i have problem, my date and time is to old
```
[root@pikvm kvmd-webterm]# date
Wed Jul 28 04:16:44 MSK 2021
```
how i can update it from NTP? also it will be useful to auto update date and time every time when pikvm connecting to network, because RPI don't have RTC clock

mdevaev, 2021-10-13:
Run `sed -i -e 's/#NTP=$/NTP=0.arch.pool.ntp.org 1.arch.pool.ntp.org 2.arch.pool.ntp.org 3.arch.pool.ntp.org/g' /etc/systemd/timesyncd.conf`, reboot and try again.