pikvm/pikvm #426: Mobile Built-in Keyboard Support

joeljacobs, 2021-08-13:
**Describe the bug**
On iOS at least, tapping inside the KVM stream window does NOT bring up the built-in keyboard. Instead, you need to use the Pikvm on-screen keyboard, which is far too large and doesn't fit on the screen with the kvm view. 

**To Reproduce**
Steps to reproduce the behavior, like:
1. Go to the KVM screen
2. Tap on the KVM stream
3. Nothing happens
4. Bring up PiKVM virtual keyboard
5. PiKVM virtual keyboard obstructs KVM stream

**Expected behavior**
Tap in the KVM stream window, and built-in iOS keyboard would come up to interact.

**Screenshots**
https://share.icloud.com/photos/0FY5cwVnaPLKKwBhk5KqTVy8Q

**Desktop:**
 - OS: iOS
 - Browser: both chrome and Safari
 - Version: iOS 13 lastest

**Pi-KVM info:**
  - RPi 4
  - Pi-KVM platform:  v2-hdmi
  - Video capture type: CSI bridge
  - KVMD version: 3.5-1
  - uStreamer version: 4.4-1
  - Linux kernel: Linux pikvm 5.10.23-2-ARCH #1 SMP Thu Mar 18 18:33:35 MSK 2021 armv7l GNU/Linux

**Additional context**
I would imagine (though I can't test) that this problem exists on Android as well

mdevaev, 2021-08-14:
Hi! The built-in keyboard is useless and does not contain all the necessary keys. I will try to find out why this is not happening.

mdevaev, 2021-10-31:
I've tested on iPad pro 2016 with iOS 14 and it's working fine. Please specify which device you use.

mdevaev, 2022-07-30:
There is no necessary information.

closed 2022-07-30 by mdevaev